cta-observatory/ctapipe #1286: Make `is_simulated` a property of the event source

MaxNoe, 2020-04-21:
I think this is important information that should not be hidden in `metadata` and be required to be implemented for all event sources.

Opinions?

vuillaut, 2020-04-21:
I agree.

watsonjj, 2020-04-21:
I have nothing against it. But we only really have one simulation source, so its quite clear already. Did you encounter a particular need for this information, other than for "metadata" reasons?

MaxNoe, 2020-04-21:
I don't really wanne see `isinstance(event_source, SimTelEventSource)` when it could be
`if event_source.is_simulated`.

watsonjj, 2020-04-21:
But what follows that `if` statement that is guaranteed to be valid for any simulation event source?

MaxNoe, 2020-04-21:
Yes, e.g. writing the mc container to the DL1 file.

watsonjj, 2020-04-21:
Okay, that sounds valid (assuming any simulation event source is guaranteed to have implemented that container)

kosack, 2020-05-04:
Sounds good to me.  You might also want some other meta info like what data level the event source provides. 

```py3
print(eventsource.is_simulated).
print(eventsource.data_levels) 
```
```
True
['R0', 'R1']
```

Where eventsource.data_level is a list of what data levels are generated: 
*  `["DL1",]` for a DL1EventSource, when that exists
*  `["R0","R1"]` for SimTelEventSource

MaxNoe, 2020-05-04:
@kosack that's a good idea, will include that

commit c8b39554 (2020-05-11): Add more properties to EventSource, implement for SimTelEventSource, fixes #1286  (#1313)

* Add more properties to EventSource, implement for SimTelEventSource

For all EventSources:
* obs_id
* is_simulation
* data_levels

For SimTelEventSource:
* mc_header

* Remove unused and not working sources in ctapipe/io/sources.py

* Use an enum for data level

* Add new properties to ToyEventSource and test source

* Split up simtel event source filling for better readability

* Reanem DL1a/b to _IMAGES/_PARAMETERS

* Use staticmethod where appropriate in simteleventsource

* Add missing import

* Ma [... cut]
closed 2020-05-11 by kosack