cta-observatory/ctapipe #1287: Avoid instantiating container defaults where possible

MaxNoe, 2020-04-21:
In the codebase, quite a few times there is code like this:

```
container = Container()
container.foo = 5
container.bar = 10
```

This is inefficient – especially for complex defaults – as the defaults are created and directly overwritten.

This should be replaced everywhere by:

```
container = Container(
    foo=5,
    bar=10,
)
```

You get the idea.

maxnoe, 2020-11-20:
Adding some debug logs and running the stage one tool with 1000 events, I got this:

```
4540x Deepcopying field true_parameters of container <class 'ctapipe.containers.SimulatedCameraContainer'>
4540x Deepcopying field azimuth of container <class 'ctapipe.containers.TelescopePointingContainer'>
4540x Deepcopying field altitude of container <class 'ctapipe.containers.TelescopePointingContainer'>
4540x Deepcopying field waveform of container <class 'ctapipe.containers.R0CameraContainer'>
4540x Deepcopying field selected_gain_channel of container <class 'ctapipe.containers.R1CameraContainer'>
4540x Deepcopying field waveform of container <class 'ctapipe.containers.R1CameraContainer'>
4540x Deepcopying field selected_gain_channel of container <class 'ctapipe.containers.DL0CameraContainer'>
4540x Deepcopying field waveform of container <class 'ctapipe.containers.DL0CameraContainer'>
4540x Deepcopying field image of container <class 'ctapipe.containers.DL1CameraContainer'>
4540x Deepcopying field image_mask of container <class 'ctapipe.containers.DL1CameraContainer'>
4540x Deepcopying field peak_time of container <class 'ctapipe.containers.DL1CameraContainer'>
4540x Deepcopying field parameters of container <class 'ctapipe.containers.DL1CameraContainer'>
4540x Deepcopying field deviation of container <class 'ctapipe.containers.TimingParametersContainer'>
4540x Deepcopying field slope of container <class 'ctapipe.containers.TimingParametersContainer'>
4540x Deepcopying field intercept of container <class 'ctapipe.containers.TimingParametersContainer'>
4540x Deepcopying field std of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
4540x Deepcopying field mean of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
4540x Deepcopying field max of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
4540x Deepcopying field skewness of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
4540x Deepcopying field kurtosis of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
4540x Deepcopying field min of container <class 'ctapipe.containers.PeakTimeStatisticsContainer'>
1170x Deepcopying field timing of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field hillas of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field intensity_statistics of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field leakage of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field peak_time_statistics of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field concentration of container <class 'ctapipe.containers.ImageParametersContainer'>
1170x Deepcopying field morphology of container <class 'ctapipe.containers.ImageParametersContainer'>
19x Deepcopying field n_trigger_pixels of container <class 'ctapipe.containers.TelescopeTriggerContainer'>
19x Deepcopying field time of container <class 'ctapipe.containers.TelescopeTriggerContainer'>
19x Deepcopying field trigger_pixels of container <class 'ctapipe.containers.TelescopeTriggerContainer'>
19x Deepcopying field dl1 of container <class 'ctapipe.containers.EventCameraCalibrationContainer'>
1x Deepcopying field trigger of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field count of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field r0 of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field dl0 of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field mon of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field calibration of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field dl2 of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field pointing of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field simulation of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field r1 of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field dl1 of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field index of container <class 'ctapipe.containers.ArrayEventContainer'>
1x Deepcopying field obs_id of container <class 'ctapipe.io.dl1writer.DL1Writer._write_simulation_configuration.<locals>.ExtraSimInfo'>
```

maxnoe, 2020-11-20:
Will try to remove those now.

maxnoe, 2023-01-04:
This was addressed for the most part in #1991 which vastly improved the performance of this