cta-observatory/ctapipe #1289: Is enum_trait duplication from `TypeTrait`?

MaxNoe, 2020-04-22:
It seems to me, that our `enum_trait` function is a duplication of traitlets buildin `Type`. 

https://traitlets.readthedocs.io/en/stable/trait_types.html#traitlets.Type

Is there a reason why this was reimplemented? Or did we just overlook the `Type` trait?

watsonjj, 2020-04-22:
Does it also detect subclasses of subclasses (etc.)? That is the special thing provided by `non_abstract_children`

watsonjj, 2020-04-22:
Also does it detect if the subclass is abstract?

MaxNoe, 2020-04-22:
So it doesn't offer options at all. It just makes sure whatever is given is a a subclass of the given klass.

And you have to give the fully qualified name.

I'll close this, as the enum thing is clearly superior for us.

closed 2020-04-22 by MaxNoe