cta-observatory/ctapipe #1293: Remove `plotting.`  `EventViewer` / `CameraPlotter` / `NominalPlotter`

MaxNoe, 2020-04-27:
Untested and currently also broken (e.g. undefined variables in the EventViewer):

The code seems not very general and wraps around the Camera and Array display in `ctapipe.visualization`.

Also, why do we have separate subpackages `visualization` and `plotting`?

```
$ pyflakes ctapipe/plotting/event_viewer.py 
ctapipe/plotting/event_viewer.py:115: undefined name 'array'
ctapipe/plotting/event_viewer.py:117: undefined name 'array'
ctapipe/plotting/event_viewer.py:121: undefined name 'array'
ctapipe/plotting/event_viewer.py:128: undefined name 'array'
```

kosack, 2020-04-27:
> Also, why do we have separate subpackages visualization and plotting?

* visualization are low-level widgets for displaying a camera or array, etc
* plotting was higher-level tools/applications that put them together into a full application 

The event viewer still works fine for me (run `ctapipe-event-viewer`), though it's a bit slow (for some reason). 

You can now do really nice Bokeh displays inline in a Jupyter notebook (see e.g. #1247) with no application or server needed, so  we could even re-produce the full capabilities of the current ctapipe-event-viewer.  If so, we could just turn it into a nice "widget" for event displaying in notebooks only.

MaxNoe, 2020-04-27:
> The event viewer still works fine for me (run ctapipe-event-viewer), though it's a bit slow (for some reason).

That's something different. This entry point call `ctapipe.tools.bokeh.file_viewer`. 

I'm talking about `ctapipe.plotting.event_viewer`

MaxNoe, 2020-04-27:
The files are `ctapipe/plotting/event_viewer.py`, `ctapipe/plotting/camera.py`, `ctapipe/plotting/array.py`

kosack, 2020-04-27:
Ah, sorry I confused the two: I've never used `event_viewer`. It seems to be stale code that can be removed (though perhaps @ParsonsRD is still using it?).  

We do need to make some nicer combined widgets that do common things like easily show an event with an array display for the ground position next to some sort of hillas-parameter display on the sky for direction (combining multiple cameras).  But probably we should start fresh and do it properly.

MaxNoe, 2020-04-27:
@kosack agreed

ParsonsRD, 2020-04-27:
Feel free to get rid of this code, it might be used for test plots in ImPACT but is not so important. It was just a very early example of how to do this kind of array plotting.

closed 2020-04-30 by MaxNoe