cta-observatory/ctapipe #1307: CameraFrame arguments in muon diagnostic plots

mgaug, 2020-05-05:
While testing the muon diagnostic plots recently, I realized that the CameraFrame constructor seems to require an argument "telescope_pointing" now, in lines 148 and 164 of muon_diagnostic_plots.py.

kosack, 2020-05-05:
It's not required, only if you want to transform those coordinates to the sky.  If you provide it, you can do the transform, if not, you cannot.

MaxNoe, 2020-05-05:
@mgaug these plotting functions have been removed, as they were untested and unmaintained for quite a while now.

I will be working with a Master student on ctapipe/LST muon analysis and bring the muon analysis here back up to speed with the ctapipe developements.

See #1261, #1297

closed 2020-05-05 by MaxNoe