cta-observatory/ctapipe #1327: Divide by zero in hillas parameters

MaxNoe, 2020-05-14:
During running the dl1 tool I noticed a divide by 0 error in the hillas parameters when calculating psi.

This should not happen, as the angle can still be well defined for `x = 0` if `y != 0`

MaxNoe, 2020-05-14:
Okay, the result is still correct though:

``` 
In [4]: np.arctan(np.inf)                                                                                                                     
Out[4]: 1.5707963267948966
```

so it would just be getting rid of the warning, which is not critical I think

kosack, 2020-05-14:
We could filter the warning, so users aren't confused. Warnings should mean something, if not people ignore all of them.

commit 422df5ee (2020-05-15): Fix hillas parameters divide by 0 warning, fixes  #1327 (#1328)

* Fix hillas parameters divide by 0 warning

* Fix concentration warning for width=0, test no warnings are raised

* Use float, fixes hdf writer
closed 2020-05-15 by MaxNoe