cta-observatory/ctapipe #1331: Failing test after clean installation of latest development version

HealthyPear, 2020-05-15:
I cleaned up my environment and re-installed the very latest version of the master branch (few minutes ago).

Installation goes ok, but one test fails and I am not sure if it's a real problem of something related to my setup.

```
______________________________________ ERROR collecting ctapipe/coordinates/tests/test_coordinates.py _______________________________________
ctapipe/coordinates/tests/test_coordinates.py:7: in <module>
    location = EarthLocation.of_site('Roque de los Muchachos')
../../anaconda3/envs/cta-dev/lib/python3.7/site-packages/astropy/coordinates/earth.py:351: in of_site
    registry = cls._get_site_registry()
../../anaconda3/envs/cta-dev/lib/python3.7/site-packages/astropy/coordinates/earth.py:532: in _get_site_registry
    reg = get_downloaded_sites()
../../anaconda3/envs/cta-dev/lib/python3.7/site-packages/astropy/coordinates/sites.py:139: in get_downloaded_sites
    jsondb = json.loads(content)
../../anaconda3/envs/cta-dev/lib/python3.7/json/__init__.py:348: in loads
    return _default_decoder.decode(s)
../../anaconda3/envs/cta-dev/lib/python3.7/json/decoder.py:337: in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
../../anaconda3/envs/cta-dev/lib/python3.7/json/decoder.py:355: in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
E   json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
========================================================== short test summary info ==========================================================
ERROR ctapipe/coordinates/tests/test_coordinates.py - json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

MaxNoe, 2020-05-15:
This seems to be a network issue downloading astropy data, not anything related to changes in ctapipe

HealthyPear, 2020-05-15:
Ah ok, good to know!

Also the other errors and warnings not related to astropy are due to this?
You can see them in the log I attached

HealthyPear, 2020-05-15:
I think some of those are real problems, because some of the examples fail, e.g.:

- `array_display.py` crashes with an `astropy.units.core.UnitConversionError`,
- `calc_pedestals.py` crashes with `ImportError: cannot import name 'pedestals' from 'ctapipe.calib'`,
- `camera_rotation.py` doesn't show anything, but I noticed that is just because a final `plt.show()` is missing,
- `coordinate_transformations.py` crashes with a `ValueError`,
- `display_event.py` doesn't work and when asked to display an event it crashes with `AttributeError: 'R0CameraContainer' object has no attribute 'image'`,
- `plot_array_hillas.py` seems to work, it just throws some warnings because some of the attributes of the coordinate frames are not specified,
- `plot_showers_in_nominal.py` same as above,
- `plot_theta_square.py` I think it works, but it finds an event with `width=0` and crashes (maybe related to the discussion of yesterday on Slack?),
- `simple_event_writer.py` is missing the input file, maybe we can use the test one from ctapipe-extra?
- `simple_pipeline.py` same as above,
- `stereo_reconstruction.py` crashes due to an `IndexError` when applying the image cleaning mask

MaxNoe, 2020-05-15:
There are some things in the output that got changed. But I don't see any problems with a fresh install on the current master (and travis succeeds).

Do you really have a fresh install / checkout? No additional files / changes? Did you run make clean before building the docs? Or `pip install -e . `?

MaxNoe, 2020-05-15:
I can reproduce some errors with the examples

HealthyPear, 2020-05-15:
Ok, cleaning the docs again helped eliminating a lot of the bulk stuff.
For the rest, I followed religiously the instrucions :)

Now the only errors remaining are 
- the `JSONDecodeError` from `astropy.coordinates` (the shows up both from `pytest` and in `tutorials/coordinates_example.ipynb`),
- an error in `atexit._run_exitfuncs:les/table_writer_reader`, specifically `AttributeError: 'NoneType' object has no attribute 'send_multipart'`.

I didn't try to rerun all the examples, but cleaning the docs cannot influence the examples

MaxNoe, 2020-05-15:
Fixes to all examples you mentioned in #1332 

Seems like the astropy site catalog is broken, can you try to remove the astropy cache so it's downloaded again?

`rm -r ~/.astropy/cache`

The other error is happening since quite some time, but does not affect anythging. I couldn't find out how to get rid of this.

HealthyPear, 2020-05-15:
Ok, I confirm that cleaning the cache of `astropy` solves that error in both pytest and docs.
The warning `WARNING: IERSStaleWarning: leap-second file is expired. [astropy.utils.iers.iers]` when compiling the docs is still there, though (probably it has a different origin).

Now pytest ends correctly, only 2 caveats (probably not urgent for now):
- the following 2 tests are skipped,

`ctapipe/reco/tests/test_ImPACT.py::TestImPACT::test_image_prediction SKIPPED                                                          
ctapipe/reco/tests/test_ImPACT.py::TestImPACT::test_likelihood SKIPPED`

- there are 91 warnings which  if you want I can share here (but I think you can see the same)

Is there an open issue about that old error? If not, I think we can open one and close this one.

HealthyPear, 2020-05-15:
I'll write you about the examples in the PR

MaxNoe, 2020-05-15:
The impact tests are skipped due to missing large test files (the image templates) this is known.

Yes, we currently have a lot of warnings. It would be great to get rid of some or all of them. 

Some come from still using prod2 files in the testing (unknown pixel shape), some from the missing obstime / location for coordinate conversions, these could actually be just ignored, as they by design that way.

HealthyPear, 2020-05-15:
> The impact tests are skipped due to missing large test files (the image templates) this is known.
> 

OK

> Yes, we currently have a lot of warnings. It would be great to get rid of some or all of them.
>
> Some come from still using prod2 files in the testing (unknown pixel shape), some from the missing [...]

Should I open a new issue about this? Just as a reference?
I can paste the output there if needed.

MaxNoe, 2020-05-15:
No need to paste the output, they are in every travis log.

HealthyPear, 2020-05-15:
Ok

Do you agree to close this?

commit dccb86b8 (2020-05-18): Fix examples, fixes #1331 (#1332)

* Fix missing unit in array display

* Fix import in pedestal example

* Add show to camera rotation

* Fix coordinate names in coordinate examples

* Remove display_event example, use ctapipe-display-dl1

* Ignore missing frame attribute warning in examples

* Ignore events with width=0 in theta square example

* Fix simple event writer, remove use of CutFlow

* Remove simple pipeline example

* Fix stereo reco example

* Skip telescopes without waveforms in pedestal example
closed 2020-05-15 by MaxNoe