cta-observatory/ctapipe #1335: Update ImPACT code

MaxNoe, 2020-05-15:
ImPACT is rather outdated with respect to the rest of ctapipe.

Some points come to mind:

* TelescopeParameter for the template paths
* Unit tests
* How to produce templates? E.g. adapt to use the stage1 tool?

kosack, 2020-05-15:
I think @ParsonsRD had a script to make templates, but it is so far not included (should be), as well as a dummy template for testing.   Who should we put as maintainer for this?  I would like to keep it around, since it's a very useful algorithm and likely will be the standard one used for final products, but it needs somebody to keep it working or at least enough of a push to get it to a testable state so others can make sure it works without breaking things.

kosack, 2020-05-15:
we can bring this up in the next ASWG meeting

MaxNoe, 2020-05-15:
We have two new PHD students, I could ask if we might be interested in template based analysis

kosack, 2020-05-15:
note there is still this PR to finish: it's been stalled for a while: #1170