cta-observatory/ctapipe #1337: wrong photo_electron_image when IGNORE_TELESCOPEs option was used running sim_telarray

yrenier, 2020-05-18:
I have corsika files with 7 telescopes and used IGNORE_TELESCOPEs=1,2,7 to keep only the LSTs (telescopes 3, 4, 5 and 6) during the sim_telarray simulations.
using ctapipe.io.event_source() on the resulting simtel file results with wrong data in event.mc.tel[tel_id].photo_electron_image.
this is due to the folowing existing code:
```
tel_index = self.file_.header['tel_id'].tolist().index(tel_id)
...
mc.photo_electron_image = (
                    array_event
                    .get('photoelectrons', {})
                    .get(tel_index, {})
                    .get('photoelectrons', np.zeros(n_pixels, dtype='float32'))
                )
```
in my case I have 'self.file_.header['tel_id'].tolist() = [3,4,5,6]' so  tel_index=1 looking for tel_id=4.
However array_event['photoelectrons'].keys() is always a subset of [2, 3, 4, 5] so the the code does not fill any info for telescopes 3 and 4 while get the info from the wrong telescopes otherwise.

The solution seems to be to use the following code instead:
```
mc.photo_electron_image = (
                    array_event
                    .get('photoelectrons', {})
                    .get(tel_id - 1, {})
                    .get('photoelectrons', np.zeros(n_pixels, dtype='float32'))
                )
```

yrenier, 2020-05-18:
The code above is for ctapipe 0.7, as the naming changed in master I will prepare a PR

maxnoe, 2020-10-26:
Pr with fix was merged.

closed 2020-10-26 by maxnoe