cta-observatory/ctapipe #1340: vitables as a dependency

HealthyPear, 2020-05-18:
is there a reason why we are not adding _vitables_ as a dependency to read the products of the writer scripts?

Are there other candidates and we leave this choice to the (low-level)-users?

MaxNoe, 2020-05-18:
Why should we add a command line utility to look into files as a dependency? It's not.

HealthyPear, 2020-05-18:
Yes sorry it's not a dependency...
I meant into the environment file

kosack, 2020-05-18:
actually, it's not a bad idea.  Originally I did not do it since vitables conflicted with many packages, but that's been improved recently, so we could add it for the release, so people have an environment out-of-the-box that can generate and view vitables (same reason we have pandas in there, in fact, even though it's not a dep)

PR #1344 (2020-05-20): add vitables to user environment
fixes #1340 , Just adds the vitables app (viewer for HDF5 files like those produced by ctapipe) to `environment.yml`

**Rationale**: we currently ask users & developers to set up their environment using environment.yml (rather than just `conda install ctapipe`), which includes packages that go beyond the required ctapipe dependencies, adding useful utilities like pandas and jupyter (also needed for documentation building).  This env will likely also be published on anaconda.org (as ctapipe-0.8-env or similar), for easy user setup.

In the future, we might want  [... cut]
closed 2020-05-20 by MaxNoe