cta-observatory/ctapipe #1341: Mean pe time for simulated events

MaxNoe, 2020-05-19:
simtel array also delivers the individual photon arrival times in each pixel.

I calculated mean arrival time in each pixel from this. The results are looking good.

However as these are lists of lists, calculating the mean in each pixel for each camera is kinda slow (1ms per telescope I found).

I see two possible solutions, calculating the mean time already in the cython part of eventio (would be easy, but right now, eventio does not really do things like this) or using something more suited to variable length arrays, e.g.: https://github.com/scikit-hep/awkward-array

Which might be a good idea anyways (maybe directly in eventio) as many things in eventio are variable length

MaxNoe, 2020-05-19:
[image]

kosack, 2020-05-19:
how about numba? We use that already, and it should be relatively fast for list looping

MaxNoe, 2020-05-19:
```
                    mc.mean_pe_time = np.array([
                        np.mean(t) if len(t) > 0 else np.nan
                        for t in pe["time"]
                    ]).astype('float32')
```

kosack, 2020-05-19:
I think if you make that a function with @jit, it would work fine - numba unrolls np.mean and I think supports list comprehensions, so not much change needed.

watsonjj, 2020-05-19:
> simtel array also delivers the individual photon arrival times in each pixel

Is this always delivered, or is it a special option that needs to be activated when running simtelarray and has additional overhead?

MaxNoe, 2020-05-19:
> Is this always delivered, or is it a special option that needs to be activated when running simtelarray and has additional overhead?

When the true image is there, also the times are there.

But yes, this is a special option that needs to be activated.

MaxNoe, 2020-05-19:
I tried with numba, but it chokes on the empty lists. awkward is slower, as it also as to first create the list of lists to an awkwardarray and that's basically uses the same loop.

MaxNoe, 2020-05-19:
I added calculation of the mean time in pyeventio here:
https://github.com/cta-observatory/pyeventio/pull/204/commits/864f193b1a413d300e2a191e58c5a13381815f93

MaxNoe, 2020-05-20:
eventio 1.2 is now out with mean_time in the output for this.

watsonjj, 2020-05-23:
Another option, if you wanted to avoid this calculation of mean_time in eventio, is to store the individual time value information in two 1d arrays, instead of the "list of list" structure. One array is the pixel each photoelectron arrives in, the other array is the time each photoelectron arrives in. Then mean_time can be calculated as:

```python
        sum_ = np.zeros(n_pixels)
        n = np.zeros(n_pixels)
        np.add.at(sum_, pixel_array, time_array)
        np.add.at(n, self.pixel, 1)
        mean_time = np.divide(sum_, n, out=np.full_like(sum_, np.nan), where=n != 0)
```

I find having the individual photoelectron times in these two 1d arrays is also more convenient

watsonjj, 2020-05-23:
Opened a PR in eventio if you are inclined to go in this direction:
https://github.com/cta-observatory/pyeventio/pull/207