cta-observatory/ctapipe #1347: stage1 performance improvements

MaxNoe, 2020-05-20:


kosack, 2020-05-21:
no description here, but I guess this a duplicate of  #1077

MaxNoe, 2020-05-21:
Not really, numba can be a way.

Sorry, I just opened this yesterday without directly writing a proper description.

I wanted to start posting profiling results and start discussing how to best approach thia.

MaxNoe, 2020-05-22:
So i'll follow this up with some performance measurements.

What really sticks out to my last check: the siegel-slopes cut the event rate in half.

kosack, 2020-05-25:
> siegel-slopes cut the event rate in half

That's unfortunate, as we definitely need a robust linear fit.  Of course there are other ways to do robust fitting, and it's not too complicated to implement  - it was just easiest to use a pre-built method.

MaxNoe, 2020-05-25:
Yes, I did and no I don't think it's because of the scipy version.

MaxNoe, 2020-05-25:
This is with the new number of islands in #1354:

[image]

kosack, 2020-05-25:
thielslopes is 2x faster (still not great, as for me siegelslopes takes 8% of the total cpu time), but for me if gives a good slope, but poor intercept

[image]

so probably rolling our own will be the best.

MaxNoe, 2020-05-25:
Fyi, a simple linear regression implemented with numpy and evaluated with numba.njit takes 1.8 µs on my computer:

```

@njit
def linear_regression(x, y):

    cov_matrix = np.cov(x, y)
    a = cov_matrix[0, 1] / cov_matrix[0, 0]
    b = np.mean(y) - a * np.mean(x)

    return a, b
```

kosack, 2020-05-25:
the fitting isn't really the main problem, it's the outlier determination, but it's clear we can also speed up `np.polyfit` (https://gist.github.com/kadereub/9eae9cff356bb62cdbd672931e8e5ec4).  Making it robust to outliers is important, since that gave large improvements in EventDisplay (when using the 2-pass charge integration that uses time gradient info).  So here, if we use the TwoPassImageExtractor, having a robust-to-outliers timing slope is needed.

MaxNoe, 2020-05-25:
I have a version with sigma clipping that runs in 8 µs here now #1354

kosack, 2020-05-25:
by the way, a tip for anyone interested, if you want more detailed profiling analysis than SnakeViz, this works well if you have qcachegrind or kcachegrind installed:

```bash
pip install pyprof2calltree
pyprof2calltree -i stage1.prof -k
```

[image]

kosack, 2020-05-25:
> I have a version with sigma clipping that runs in 8 µs here now #1354

that should be enough then

maxnoe, 2020-10-26:
Fixed by merging #1419 for now, sure there is plenty of optimizations left to do but the immediate issue is fixed I would say.

closed 2020-10-26 by maxnoe