cta-observatory/ctapipe #1352: The doc-string of this function looks wrong & some general doc ideas.

dneise, 2020-05-21:
Just stumbled over this function looking for something else. 
No idea if it is ever used or if it is important, that's also not the point of this issue.

https://github.com/cta-observatory/ctapipe/blob/23bf35655f9d6bd0215dba024cadc09187c2ac40/ctapipe/tools/stage1.py#L426-L442

The parameter names in the doc-string do not fit to the parameter names of the function.

In this context: I think there are some tools, which might be interesting, if ctapipe would like to have amazing documentation. 

 - [pyment](https://github.com/dadadel/pyment) can auto-generate doc-strings (in different styles, *numpydoc* is included) for code, which happens to be poorly documented. 
 - [pyenchant](https://github.com/pyenchant/pyenchant) can in combination with pylint be used to spellcheck code. (a grammar checker would be nice to, but hey .. it's a start). c.f. [SO question](https://stackoverflow.com/q/27162315/11635405)
 - [pydocstyle](https://github.com/PyCQA/pydocstyle) can check if doc-strings adhere to certain styles, e.g. to out preferred numpydoc style. 

Unfortunately I found no tool yet, which was able to check for "doc-string vs function parameter list consistency". 

In principle it should be possible to check for parameter names and even types in case of type annotations automatically and alert devs in case doc-strings were accidentally not updated along with the code. 

I personally would not claim, that these tools need to be automatically applied to the code base at this moment. Since there will be a lot of false positives and there will be a learning curve. 
But maybe one could advertise the usage of (some of) these tools, so devs might use them before opening a PR, and thus maybe improving the doc quality a lot.

kosack, 2020-05-21:
I use pycharm, which does it automatically. We just forgot to update this when refactoring.  I wouldn't worry though, as this is a private function, and this whole script will be refactored into components (e.g. ImageProcessor and DL1Writer) for the next minor release, so all will be re-worked.

kosack, 2020-05-21:
Use use pydocstyle in codacy, but right now codacy is a bit broken for me (sometimes it uses python-3 sometimes python-2, as if 2 instances are running).  So once that's fixed, it should be checked better.   I didn't know about pyenchant, but we could add that to future builds as well.  PyCharm does spellchecking, but of course that requires the dev to run the check.

kosack, 2020-05-21:
> Unfortunately I found no tool yet, which was able to check for "doc-string vs function parameter list consistency".

Actually, now that I look, PyCharm does warn about this and even underlines the wrong keywords in the docstring.  However, you can't run that in a CI system

maxnoe, 2023-01-04:
```
❯ pydocstyle --convention=numpy ctapipe | wc -l
4060
```

:cry:

HealthyPear, 2023-07-24:
If you want my two cents, I have experimented quite a bit in the last period on SWGO software (which is currently wildly developed, so it's quite a stress-test...)

- [numpydoc](https://numpydoc.readthedocs.io/en/latest/index.html) is actually its own package, which can be used as a sphinx extension and has its own validator; it can be used from the sphinx-build and soon even as a pre-commit hook; the only drawback (on top of some smaller negligible things) is that it doesn't support type annotations, which is something I saw popping out in #1775
- [napoleon](https://www.sphinx-doc.org/en/master/usage/extensions/napoleon.html) is a built-in sphinx extension that plays well with numpy-style docstrings and type annotations, but doesn't have a validator

maxnoe, 2023-10-21:
We should switch from flake8 to ruff and also enable the docstring linting