cta-observatory/ctapipe #1355: Image unused in timing parameters

MaxNoe, 2020-05-25:
Currently, the image is not used in the determination of the timing parameters, although the function takes it.

I suspect, weighting was planned but never implemented?

Do we want to weight the linear regression with the image / sqrt(image)?

kosack, 2020-05-25:
it should improve the fit. At least I can't see a major downside. Perhaps it should be an option, on by default, in case we want to test without it (`weighted=True`)

calispac, 2026-09-29:
I recently looked into the performance of `timing_parameters` and realized that the current implementation is relatively expensive. The main reason seems to be the use of Least Trimmed Squares (LTS) regression: the linear regression is performed iteratively on randomly selected subsets, which can result in up to ~100 iterations in the worst case.

The current implementation uses Numba to make this iterative procedure reasonably fast, but this also makes the implementation harder to follow and maintain. Since this is essentially a linear regression problem, I wonder whether we could use a more standard regression implementation instead.

I understand the motivation for using LTS here: it makes the timing-parameter estimation more robust by selecting the subset of points with the smallest residuals, thereby reducing the influence of outliers.

However, I wonder whether a weighted least-squares approach might be more appropriate for this particular application. The peak time of a pixel is not necessarily equally well determined for all pixels, and the image intensity provides information about the statistical quality of the timing measurement. Weighting the pixel times according to their intensity could therefore provide a more direct way of accounting for this.

In particular, extending

`ctapipe.fitting.linear_regression(X, y, weights=None)`

to support weighted least squares should be relatively straightforward. The implementation could follow the standard formulation of Weighted Least Squares:

https://en.wikipedia.org/wiki/Weighted_least_squares

This would also avoid the iterative random-subsampling procedure used by LTS, potentially making the timing-parameter calculation both simpler and faster.

I think it would be worth comparing the two approaches on representative data before changing the current implementation: LTS versus intensity-weighted least squares, looking at both the resulting timing parameters and the computational cost.

calispac, 2026-09-29:
#1645

maxnoe, 2026-09-29:
>  Since this is essentially a linear regression problem, I wonder whether we could use a more standard regression implementation instead.

We started with this actually and replaced it with the LTS after some discussion, especially the comparison with CTA-Mars analysis.

maxnoe, 2026-09-29:
Original PR exchanging `scipy.stats.siegelslopes` with the LTS regression:  https://github.com/cta-observatory/ctapipe/pull/1419

Previous replacement of simple `np.polyfit` with `siegelslopes`: https://github.com/cta-observatory/ctapipe/pull/1351

calispac, 2026-09-29:
Thanks for the quick reply and finding the exact PRs for which the changed happened. 

Maybe a solution could be to allow to configure samples=20, relative_sample_size=1, max_iterations=1, eps=1e-12 parametere of the lts_linear_regression from the ImageProcessor such no iteration are performed. ?

maxnoe, 2026-09-29:
Sure, exposing configuration options would be fine