cta-observatory/ctapipe #1360: HDF5TableReader does not support prefixed columns

LukasNickel, 2020-06-08:
The stage1 tool collects all parameters from the camera images in one ImageParametersContainer.
While writing, the containers get expanded and prefixes get added to avoid collisions. So (x,y,...) from the HillasParametersContainer maps to (hillas_x, hillas_y, ...) and so forth.
See 
https://github.com/cta-observatory/ctapipe/blob/8c02fb772ec54c200b69721baf9ef20f590cd749/ctapipe/tools/stage1.py#L606-L609
 and 
https://github.com/cta-observatory/ctapipe/blob/8c02fb772ec54c200b69721baf9ef20f590cd749/ctapipe/tools/stage1.py#L716

The HDF5TableReader on the other hand tries to match the column names directly to the container, see https://github.com/cta-observatory/ctapipe/blob/8c02fb772ec54c200b69721baf9ef20f590cd749/ctapipe/io/hdf5tableio.py#L399-L402
Besides not being able to directly read into the ImageParametersContainer(?), the matching also fails when trying to read into the individual containers.

The easiest fix would probably be to add a check for prefixes to the read() Method of the HDF5TableReader.

LukasNickel, 2020-06-08:
Playing around with this I see multiple ways to approach the prefix issue:

- Add an option to the `read()` method 
- Add a property to the reader, so `read` can check `self.prefix`
- Always use the container prefix. One would then need to initialize a container and remove the prefix before using the `read` method.

kosack, 2020-06-09:
I guess the most flexible is option 1 or 2.  If you go for 3, it just means we have to be careful to not change the prefixes (which we do currently in the stage1 tool I think). I can see in particular a use case that wouldn't work in option 3, namely if we write for example two sets of hillas parameters with different cleaning to the same table (with different prefixes). 

I guess for 1 it would look like this, which isn't too bad. There could however be an option to use the default prefix in the container (somecontainer.container_prefix), e.g if prefix=None use the container prefix, if "" it would be no prefix

```py3
reader.read(table_name=table, container=hillas, prefix="hillas")
reader.read(table_name=table, container=index, prefix="")
```

LukasNickel, 2020-06-09:
> I guess for 1 it would look like this, which isn't too bad. There could however be an option to use the default prefix in the container (somecontainer.container_prefix), e.g if prefix=None use the container prefix, if "" it would be no prefix

I think thats a good way to do it. It feels slightly hacky to differentiate between None and an empty string, but it's not like this is a common use case for the end user.

PR #1364 (2020-06-30): Add support to read from prefixed columns
This fixes #1360 in order to work towards a DL1EventSource.

- The `HDF5TableReader.read()` method now has an additional, optional argument `prefix`
- Following the discussion in #1360 three use cases are supported:
    - `prefix=False`: No prefix. This is the default, because per default the writer does not add a prefix as well
    - `prefix=True`: Using the container prefix
    - `prefix="my_prefix"`: Using a custom prefix

This enables the reading of the image parameter containers, that are written by the stage1 tool, but support for reading into mu [... cut]
closed 2020-06-30 by kosack