cta-observatory/ctapipe #1367: HDF5TableReader can not setup multiple containers for one table

LukasNickel, 2020-06-17:
(Following up the discussion in #1357)
This is already mentioned in the docstring of the HDF5TableReader, but will probably need to be solved in order to get the DL1EventSource working (without hacks).

Currently the `read()` method takes only one container and proceeds to then map the table columns to the container columns. After that, `_cols_to_read` contains only the columns that are in the table and match the container fields. This is tied to the table however, a second call to `read()` with another container will not work, because `_cols_to_read` still contains the columns matching the first container and does not get updated.

Example (requiring #1364 in order to read the dl1 table):
[image]

This means, that right now the only ways to read from the parameters table in a dl1 file would be:
- Manually reset `_cols_to_read` every time you want to iterate over any of the generators
- Setup multiple readers, one per container class

If we still want that 1 to 1 mapping of columns to fields in order to avoid the overhead of testing each column every time, we could maybe put an additional layer to the `_cols_to_read` dictionary. Right now it is constructed as `{table_name: [list_of_columns]}`. It could be expanded to `{table_name: {container_name/prefix name: [list_of_columns]}}` or smth similar.

Or we do not support multiple calls to `read()` on the same table at all and require a list of all containers to be handed on the single call. Potentially also in a second function like `read_multiple()` to make sure `read()` still works as expected everywhere.

Part of the question probably is whether we want multiple containers in one table, but without prefixes. The parameters table contains prefixes everywhere, but there might be reasons to write multiple containers with overlapping columns (like event _id ) leading to naming conflicts.

kosack, 2020-06-19:
This comes from the fact that we refactored TableWriter to support multiple containers, but didn't fix TableReader in the same way.  I think the best thing would be just to do the same as in TableWriter, and accept a list, and loop over it when building the scheme.

kosack, 2020-06-19:
Another way I had originally thought to implement this in both Reader and Writer classes was to just make a CompoundContainer class that looks like a single container, but contains a list of them (then the logic is in there, not in the Reader/Writer), but I'm not sure that ends up being simpler.

LukasNickel, 2020-06-19:
> Another way I had originally thought to implement this in both Reader and Writer classes was to just make a CompoundContainer class that looks like a single container externally, but contains a list of them (then the logic is in there, not in the Reader/Writer, which only operates on one container at once), but I'm not sure that ends up being simpler.

Thats more or less what the ImageParametersContainer does, right? Except the structure is not flat, so the Writer/Reader would need some adaptions.

closed 2020-08-06 by LukasNickel