cta-observatory/ctapipe #1368: Missing container fields for the DL1EventSource

LukasNickel, 2020-06-19:
While working on the DL1EventSource, I figured that there is some information, that we might want to provide, that is not currently supported by the DataContainer. Namely:

- The DL1Container is missing a `tels_with_data`field, that is present in the lower data levels.
- There is no field for the MCDL1Containers. These are written out by the stage1 tool and include the true images and true parameters (but no true peak time or true image mask). Going with the same structure as with the measured dl1 information, we might want to add a MCDL1Container, that maps the telescope ids, and add a field mc_dl1 to the DataContainer.

Is there any reason to proceed in a different way? Per shower and per run mc information is already stored in the DataContainer anyway.

MaxNoe, 2020-06-19:
> The DL1Container is missing a tels_with_datafield, that is present in the lower data levels.

I think we said some time ago that these are redundant with the keys of the telescope maps

LukasNickel, 2020-06-19:
Agreed, but that reasoning applies to lower levels as well, no?

kosack, 2020-06-19:
There is still a lot of container structure to clean up, so I'm not surprised.  However, for the tels_with_data, outside the Trigger table I think Max is right that that is really redundant at this point.  So you only need to read back into the trigger table, and ensure the tel[] maps have the right data. 

For the MCDL1 parameters, the peak time was added only recently, so we don't write it out (only the true image so far). We do write the MCDL1Containers, that column is only missing when it is not filled.

kosack, 2020-06-19:
Actually, I just noticed that event/subarray/tels_with_triggger is never written out... it used to work, so I think it's a bug in the stage1.py script.

kosack, 2020-06-19:
removing the exclude line (670 above) fixes it:
[image]

MaxNoe, 2020-06-19:
You should fix the exclude method to not only do full matches or have a `regex=True` argument

LukasNickel, 2020-06-26:
Do we want to include code, that reads the "broken" dl1 files or just assume files have been written after this bug fix to the stage1 tool? Or alternatively "fix" the old files manually?

MaxNoe, 2020-06-26:
As there is no way to recover that information from the files, we should not attempt it.

LukasNickel, 2020-06-26:
You can perform queries on the telescope tables to look for the array event ids. Assuming that no data was discarded between the trigger and the dl1 calculation that should be the missing information.

kosack, 2020-06-26:
> assume files have been written after this bug fix to the stage1 tool? Or alternatively "fix" the old files manually?

I'd just go for assuming that it's fixed. We haven't written any large-scale data with the broken version anyhow, so there's nothing really to support

LukasNickel, 2020-06-26:
Agreed. It would make the code pretty ugly. Just wanted to confirm, that there is no super important dl1 data yet.

maxnoe, 2020-10-26:
Is this still an issue @LukasNickel ?

closed 2021-02-19 by maxnoe