cta-observatory/ctapipe #1372: tels_with_trigger + allowed_tels in dl1 and simtel files

LukasNickel, 2020-07-03:
I am not 100% sure if this is a bug or a design decision I do not understand rightaway. 

If `allowed_tels` is given while calling the stage1 tool, only events and telescope trigger information of the selected events are saved (as expected).
The `tels_with_trigger` column on the other hand is a boolean mask of all telescopes in the subarray of the simtel file. First of all I would expect a list of ids instead of a mask, but more importantly I would only expect information about the processed telescopes, because afaik the dl1 file does not include an explicit mention of the telescopes selected during processing(?).
This leads to some weird behaviour when trying to iterate through the telescope trigger table, because the length of the table does not match the sum of `tels_with_trigger` information.

Example of a dl1 file with `--max-events 10 --allowed_tels=[1,2,3]` :
- Dl1 Information is only present for the selected telescopes.
- 10 Events are processed
- 23 measured images
- Many more `tels_with_trigger`
[image]

The same confusion arises with the SimtelEventSource where the trigger information does not respect the `allowed_tels`, so the issue (if we consider it one) would be located deeper.
[image]

MaxNoe, 2020-07-03:
I don't think this is a bug. Why discard this information?

And we use a boolean mask because variable length arrays are not directly supported by tables / our containers.

LukasNickel, 2020-07-03:
I see that argumentation and thats why i mentioned I'm not 100% sure.
However the simtelfile includes trigger information for every mentioned telescope and is consistent in that way. The dl1 file on the other hand mentions triggered telescopes, but discards the related telescope trigger information if the telescope is not in the allowed telescopes. So there is at least some inconsistency.

maxnoe, 2021-02-19:
Connected to #1624 and #1623

maxnoe, 2025-08-21:
Fixed in #2136

closed 2025-08-21 by maxnoe