cta-observatory/ctapipe #1373: Passing a mask to MuonIntensityFitter?

moralejo, 2020-07-07:
The calculations to estimate the light in a muon ring are better done on a subset of pixels, usually selected up to a distance relative to the fitted ring. This used to be possible in 0.7 with the now removed MuonLineIntegrate, but its replacement MuonIntensityFitter is missing this feature, as far as I can see.

https://github.com/cta-observatory/ctapipe/blob/262b6aa887a33907901d7b0cd948b9c937ec60b9/ctapipe/image/muon/intensity_fitter.py#L443

moralejo, 2020-07-07:
In the older version what was done is that the x,y arrays of pixel coordinates were passed as arguments, not read from the telescope description. In this way one could pass masked versions of  x,y and image to obtain the desired behavior. This would also be a good solution.

moralejo, 2020-07-20:
Addressed in https://github.com/cta-observatory/ctapipe/pull/1389

MaxNoe, 2020-07-20:
An issue is normally only closed when the corresponding fix is merged into master. Reopened.

moralejo, 2020-07-20:
> An issue is normally only closed when the corresponding fix is merged into master. Reopened.

That is a recipe for having duplicated discussions, I am afraid. If I had opened it directly as a PR I guess we would not have opened a parallel issue on the same topic.

commit 597c1dc3 (2020-07-22): Mask for muons, fixes #1373 (#1389)

* Use EventSourceFactory. Corrected reading of camera geometry.

* Various bug fixes and improvements in muon ring calculations. There were
several issues with units (e.g. radians & deg)

* Solved codacy issues

* Removed trailing whitespace

* Allow the use of masks in the muon intensity fits

* Made mask an optional argument
closed 2020-07-22 by kosack