cta-observatory/ctapipe #1376: How to switch off charge integration corrections?

FrancaCassol, 2020-07-10:
Hi,
there are cases in which one do not want to apply the pulse shape integration corrections to the integrated charges . With the present code, are there better methods than patching the reference pulse shape?

watsonjj, 2020-07-11:
This was discussed here:
https://github.com/cta-observatory/ctapipe/pull/1242#issuecomment-603217848

A bool trait could be added to the extractors to disable the correction...

What is the reason why you would not want to apply it? It is a powerful part of the extractor, as it removes the dependance on the integration window size and shift used. If we start allowing the correction to be skipped, then there are possible cases were calibration coefficients may be applied incorrectly to data as they have each been extracted with different integration window sizes.

moralejo, 2020-07-15:
> This was discussed here:
> [#1242 (comment)](https://github.com/cta-observatory/ctapipe/pull/1242#issuecomment-603217848)
> 
> A bool trait could be added to the extractors to disable the correction...
> 
> What is the reason why you would not want to apply it? It is a powerful part of the extractor, as it removes the dependance on the integration window size and shift used.

It is useful to know the effect of the correction, at least for tests.  It is anyway not a perfect correction, since it is done for a given assumed pulse shape, which is the one for a single p.e.

> If we start allowing the correction to be skipped, then there are possible cases were calibration coefficients may be applied incorrectly to data as they have each been extracted with different integration window sizes.
If they are,

MaxNoe, 2020-07-15:
If it is just about knowing the correction, you can call the method independently of any processing:

```
print(calib.image_extractor._calculate_correction(telid=telid))
```

FrancaCassol, 2020-07-15:
Hi @watsonjj,
sorry for my late reply I was on vacation.

> A bool trait could be added to the extractors to disable the correction...

yes, I think this would be a good solution

> What is the reason why you would not want to apply it? 

In our case we want to compare lstchain results with ctapipe 0.7 (where we do not apply charge correction on the data) and ctapipe 0.8 (again without corrections obviously). Also, as Abelardo mentioned,  it would be interesting to switch them on and off in order to study their impact on the energy resolution (not so clear to me, given that they are based on the single pe pulse shape). But, there are surely  other situations where you do not want to have them (e.g. if you do not know precisely your reference pulse shape ;-) )

commit 3e56e523 (2020-09-22): Add config options to disable integration correction, fixes #1418, fixes #1376 (#1429)

* Add config options to disable integration correction, fixes #1418, fixes #1376

* Fix typo
closed 2020-09-22 by MaxNoe