cta-observatory/ctapipe #1377: Error in optical efficiency from muon ring?

moralejo, 2020-07-12:
https://github.com/cta-observatory/ctapipe/blob/262b6aa887a33907901d7b0cd948b9c937ec60b9/ctapipe/image/muon/intensity_fitter.py#L258

I think there is a factor 0.5 missing here. the factor comes from sin^2(theta)/tan(theta) = 0.5 sin(2*theta), see e.g. eq (7) of  https://arxiv.org/abs/1907.04375
The factor  was there in previous versions of this code.

This produces an underestimation of the telescope efficiency by the same factor.

MaxNoe, 2020-07-13:
The code does not directly implement these formulas and no reference is given what exactly was implemented here. So I am at a loss.

The formula you mention has no wavelength but an energy dependence and contains a factor of `hc` which is also nowhere to be found in the code.

MaxNoe, 2020-07-13:
I think we should make sure the code is as close as possible to the paper you linked.

moralejo, 2020-07-13:
> The code does not directly implement these formulas and no reference is given what exactly was implemented here. So I am at a loss.
> 
> The formula you mention has no wavelength but an energy dependence and contains a factor of `hc` which is also nowhere to be found in the code.

Yes, it is not straightforward, but I could not access other references where things are closer to the approach implemented in ctapipe, just because they are behind paywalls for me now (working @ home).
In short: hc 'disappears' simply because the mu-Cherenkov photons energy integration is done in wavelength. 
But the factor 0.5 comes just from that simplification of the sin^2, which also appears in the paper.

MaxNoe, 2020-07-13:
> In short: hc 'disappears' simply because the mu-Cherenkov photons energy integration is done in wavelength.
But the factor 0.5 comes just from that simplification of the sin^2, which also appears in the paper.

Ah, of course. Checking with the older code, it seems like this factor of 2 was indeed lost.

I would propose to just readd it for now but also try to move to a more straight forward implementation of this or another suitable reference and clearly link to that in the code.

moralejo, 2020-07-13:
Hi, here is a more clear reference:  https://arxiv.org/pdf/1403.4550.pdf
Take eq. 3 from there. This is what we integrate in wavelength (1/lambda^2). We do not introduce any atmospheric absorption, nor do we introduce wavelength-dependent efficiency. That is all incorporated in our global "efficiency" factor. That is, in eq (3) we replace:
psi(lambda)*a(l,lambda)  =>  efficiency

commit 052fe3bd (2020-09-07): Add missing factor of 0.5 in muon efficiency fit, fixes #1377 (#1379)

* Add mission factor of 0.5 in muon efficiency fit, fixes #1377

* Add hess muon calib reference

* Fix tabs vs. space in bibliography
closed 2020-09-07 by MaxNoe