cta-observatory/ctapipe #1381: Ring intensity container missing in the muon containers

rlopezcoto, 2020-07-13:
Probably to be implemented in the `MuonParametersContainer`
https://github.com/cta-observatory/ctapipe/blob/262b6aa887a33907901d7b0cd948b9c937ec60b9/ctapipe/containers.py#L728

the `intensity` of the muon ring is missing in the container list.
Discussed in https://github.com/cta-observatory/cta-lstchain/pull/465, maybe we can consider some of the other parameters missing

moralejo, 2023-10-25:
Currently there is an "intensity_ratio", but it is computed only with pixels which survived the cleaning (which is not good for muon calculations, because a reduction in transparency would mean both less light per pixel, **and** fewer pixels after cleaning). Besides, the "ring" seems to be defined as +-0.75 pixels around the fitted circle radius, which seems a bit small (i.e. too sensitive to uncertainties in the fitted ring center, and also to possible variations of the optical PSF). I think we should use a wider ring (and count the light in all pixels, regardless of whether they survived the cleaning or not).

maxnoe, 2025-08-26:
ring_intensity was added some time ago

closed 2025-08-26 by maxnoe