cta-observatory/ctapipe #1388: CameraCalibrator always overwrites DL0

MaxNoe, 2020-07-17:
The `CameraCalibrator` always overwrites the DL0 data without a check that it is already present.

It also is a bit surprising to me, that a DataVolumeReduction is part of something called `Calibrator`. This ought to be an extra step.

watsonjj, 2020-07-17:
If you are applying `CameraCalibrator` to an event, surely the correct assumption is that you wish to overwrite any previous calibration. This is more clear for a DL0 -> DL1 transition.

Technically `DataVolumeReduction` doesn't belong in the chain - as we will be reading the DL0 data in final system. I.e. it could also become part of the EventSource... But I never made this change because it could inhibit the exploration of different data volume reduction algorithms. However, maybe it makes sense to have the DataVolumeReduction as part of `EventSource`. Then we are truly heading to the concept that we start from DL0 data in the chain...

MaxNoe, 2020-07-17:
> If you are applying CameraCalibrator to an event, surely the correct assumption is that you wish to overwrite any previous calibration. This is more clear for a DL0 -> DL1 transition.

Not really. Maybe I only want to get the one data level I do not have yet. 
I think this is also a valid use case and which one is the more sane default is probably a very individual question.

MaxNoe, 2020-07-17:
Also, I think we should also use the config system for the calibrator's subcomponents. Like this, the config doesn't have to be repeated in every tool using those.

At least providing the possibility of setting it via the config system should be there, maybe like this:

watsonjj, 2020-07-17:
> Also, I think we should also use the config system for the calibrator's subcomponents. Like this, the config doesn't have to be repeated in every tool using those.
> 
> At least providing the possibility of setting it via the config system should be there.
> 
> As it is now, we have a strange mixture of using the config system and requiring pre-configured components to be passed in, which complicates writing tools or components using those components.

Well, I think some decisions need to be made on how to handle interleaved calibration events. I discussed some approaches here https://github.com/cta-observatory/ctapipe/issues/1147. The reason it relates to this discussion is that if multiple CameraCalibrators are used (like in option 3), then having the components created via traits won't work - we can't configure the different CameraCalibrators to be different with our current configuration system.

MaxNoe, 2020-07-17:
> The reason it relates to this discussion is that if multiple CameraCalibrators are used (like in option 3), then having the components created via traits won't work - we can't configure the different CameraCalibrators to be different with our current configuration system.

I think this is not true. the cameracalibrators would be subcomponents of different classes, right?

So, configuring them with different options would work like this:

```
{
    "PedestalCalculator": {
        "CameraCalibrator": {...}
    },
    "GainCalculator": {
        "CameraCalibrator": {...},
    },
    "FlatFieldCalculator": {
        "CameraCalibrator": {...},
    }
}
```

Like in this example: 

```
from ctapipe.core import Component
from ctapipe.core.traits import Int
from traitlets.config import Config

class Sub(Component):
    value = Int(default_value=1).tag(config=True)

class A(Component):

    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self.sub = Sub(parent=self)

class B(Component):
    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self.sub = Sub(parent=self)

class Parent(Component):
    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self.a = A(parent=self)
        self.b = B(parent=self)

config = Config({
    'A': {'Sub': {'value': 2}},
    'B': {'Sub': {'value': 3}},
})

p = Parent(config=config)

print(p.a.sub.value)
print(p.b.sub.value)
```

watsonjj, 2020-07-17:
Thanks, thats good to know. Then there should be no problem.

kosack, 2020-10-07:
> It also is a bit surprising to me, that a DataVolumeReduction is part of something called Calibrator. This ought to be an extra step.

Just to comment, trace integration is also not really calibration, but I think it's fine to have it in this class - if you really want to do things separately, you can.  This is a convenience class to do the first steps of the analysis in one go.  There will be similar classes (ImageProcessor) soon that does the next steps (cleaning, parameterization) as in the stage1 tool.  So maybe it's just not a good name, but I think it's ok.  

We could consider a new naming scheme for these high-level super-components, like CameraWaveformProcessor instead of CameraCalibrator (does R1 to DL1 image) and CameraImageProcessor (does DL1 image to DL1 param), but not sure that is any clearer.

maxnoe, 2020-10-26:
The point is, that these classes should be configurable through the config system from the commandline, otherwise it's not much of a convenience If I have to reimplement the stage1 tool for a small option.

So having traitlets like `redo_dl1_calibration = Boolean(default_value=True)` would be nice to have.