cta-observatory/ctapipe #1391: pixel-intensities of dl1 after using CameraCalibrator coming out same as pixel-intensities of r1

Writasree, 2020-07-20:
After using CameraCalibrator [calib = CameraCalibrator(subarray=source1.subarray) \\ calib(event)] , the array that I'm getting with 'event..dl1.tel[1].image' is coming out to be exactly same as the r1-waveform for the corresponding telescope. The dl0 waveform should be same as that of r1 waveform because I haven't used any data volume reduction. But why is the dl1 image array same as the r1 waveform? Are they supposed to be same arrays or am I making any mistake while using CameraCalibrator?

vuillaut, 2020-07-20:
Hi.
You must do something wrong there. A complete copy of your code would help spot the error.
In any case, this works fine:

```
import ctapipe
print(ctapipe.__version__)

from ctapipe.io import event_source
from ctapipe.utils import get_dataset_path
source = event_source(get_dataset_path('gamma_test.simtel.gz'))
from ctapipe.calib import CameraCalibrator
calib = CameraCalibrator(source.subarray)
event = next(iter(source))

tel_id = list(event.r0.tels_with_data)[0]
print(event.r1.tel[tel_id].waveform)
print(event.r1.tel[tel_id].waveform.shape)
print(event.dl1.tel[tel_id].image)
calib(event)
print(event.dl1.tel[tel_id].image)
print(event.dl1.tel[tel_id].image.shape)
```

```
0.8.0
[[-0.1142795  -0.5818849   0.04158897 ... -0.27014798 -0.5818849
   0.04158897]
 [-0.02219103  0.13554978 -0.02219103 ...  0.05667938  0.2144202
  -0.17993185]
 [ 0.40713063  0.06708676  0.06708676 ... -0.15960917 -0.15960917
   0.06708676]
 ...
 [ 0.05420959 -0.03591906 -0.03591906 ... -0.306305    0.3245955
   0.14433824]
 [ 0.3222024  -0.07432787  0.22306983 ... -0.272593   -0.07432787
  -0.17346044]
 [-0.08560796  0.02801528 -0.19923119 ...  0.02801528  0.02801528
  -0.19923119]]
(2048, 25)
None
[ 2.9408867  -0.2342076   2.3965225  ... -0.30564296  0.7684283
 -0.5696783 ]
(2048,)

Writasree, 2020-07-23:
Hi,
Thank you so much for your reply. I've understood why I was getting the same pixel-intensity from both R1container and DL1container- the output format of the simtel file that I was working with had ADC sums instead of ADC samples. As n_samples was 1 for my simtel file, the dl1 calibration was not happening. 
Now I've a slight confusion. I've produced two simtel files with all other parameters being same; just in one file the output format has ADC sum and in the other file, the output format has ADC samples. The pixel intensity that I'm getting after using dl1calibration on these simtel files are not coming out to be same. So, which set of pixel intensities among these two are more practically acceptable?

MaxNoe, 2020-07-23:
These are two different things and that the output is not the same is not surprising. 
You have to use what is appropriate for the telescope you are trying to simulate.

vuillaut, 2020-07-23:
I suppose the file you produced with Simtel having the sum is simply the sum of the ADC charges over all samples.

This is completely different from a complete calibration chain that will integrate samples over a selected window and transform ADC to photo-electrons values... 

I'd happy to continue the discussion if you need further help.
However, as the issue is not related to ctapipe, I am closing it.

closed 2020-07-23 by vuillaut