cta-observatory/ctapipe #1396: Time flat fielding

MaxNoe, 2020-07-22:
The DL0→DL1 calibration should provide for the possibility to calibrate time offsets between pixels.

In the most simplest form, this would just mean subtracting them from the extracted peak times:

```
event.dl1.tel[tel_id].peak_time -= event.calibration.dl1.tel[tel_id].time_offsets
```

However, this simple approach would harm extractors using time of multiple pixels like the `NeighborPeakIntegrator`.

I would propose to add the `time_offsets` field to the `DL1CameraCalibrationContainer`, do the simple thing where it suffices but look how such offsets could be used with the `NeighborPeakIntegrator`.

MaxNoe, 2020-07-22:
Actually, the DL1CameraCalibrationContainer already has a `time_shift` attribute.

moralejo, 2020-07-22:
> 
> I would propose to add the `time_offsets` field to the `DL1CameraCalibrationContainer`, do the simple thing where it suffices but look how such offsets could be used with the `NeighborPeakIntegrator`.

That's very important for real data, I had actually forgotten that this affects the muon ring analysis in LST1, for which we are  using GlobalPeakIntegrator to avoid signal bias in pixels with no C-light. The extractors should get as input an array of time shifts (since trying to shift & re-sample the waveforms would be too complicated and unnecessary).

watsonjj, 2020-07-22:
> The extractors should get as input an array of time shifts (since trying to shift & re-sample the waveforms would be too complicated and unnecessary).

I don't think performing inside the extractor instead of shifting the waveforms avoids the complicated aspect. Either the shifts are on an integer sample level (simple) or a subsample level (complicated no matter where its performed). I'd love to be demonstrate wrong though.

For the CHEC-S calibration chain, we used the following to shift the waveforms before passing them to the charge extractors (subsample amounts, linear interpolation for resampling). 
https://github.com/sstcam/CHECLabPy/blob/master/CHECLabPy/calib/time_calibrator.py

We will investigate better approaches for the SST camera (perhaps charge resolution is better with integer sample shifts and no interpolation, or an alternative to linear interpolation for subsample shifts). It is also worth noting that we might also apply this during the R1 calibration - thereby ensuring no problems for the R1 Online Analysis, and also not requiring any further consideration in the DL0->DL1 step.

MaxNoe, 2020-07-22:
Maybe one could think about applying an integer correction before and a fractional one < 0.5 samples after peak extraction?

moralejo, 2020-07-22:
> I don't think performing inside the extractor instead of shifting the waveforms avoids the complicated aspect. Either the shifts are on an integer sample level (simple) or a subsample level [...]

I agree to that, the only complicated thing is indeed the resampling to allow for non-integer corrections. And I would prefer not to apply such a procedure to produce the DL0, but rather simply keep the pixel-wise time shift for later use e.g. in extractors. It will allow a much easier re-doing of that relative time calibration offline, I think.

maxnoe, 2020-10-14:
I want to work on this now again, since we want to move https://github.com/cta-observatory/ctapipe_io_lst/pull/44 forward and a correction like this is needed for lst data to achieve the same calibration quality as currently in lstchain.

I think we solved the problem of needed low-level information to *calculate* the correction and will do this inside `ctapipe_io_lst`, so the problem remaining for ctapipe is applying the correction.

For the most part this is simple enough, I think. Three points come to mind: 
the `GlobalPeakWindowSum`, `TwoPassExtractor` before extraction and all the extractors after the peak time has been extracted.

watsonjj, 2020-10-14:
> For the most part this is simple enough, I think. Three points come to mind:
the GlobalPeakWindowSum, TwoPassExtractor before extraction and all the extractors after the peak time has been extracted.

Can you explain a bit more what you propose to be implemented? I didn't understand this paragraph.

> As this shift can change per event (it does for LST), it probably needs to live in dl0.event.telescope in the file and DL0CameraContainer in the event structure.

There is a field ready to be used in the `DL1CameraCalibrationContainer` called `time_shift`
https://github.com/cta-observatory/ctapipe/blob/b2e8ca9b923b7d64be04aad7e683197279a78357/ctapipe/containers.py#L305

maxnoe, 2020-10-14:
Ah, great. The question is then if a correction to be applied during DL0 → DL1 is correct in the DL1 container.

I think it would make more sense at DL0, since it is needed to create DL1.

maxnoe, 2020-10-14:
> Can you explain a bit more what you propose to be implemented? I didn't understand this paragraph.

For simple extractors, this is simple: subtract the shift from the extracted peak time.

It get's more complicated for the extractors that combine several pixel waveforms to do the actual extraction,
e.g. global peak or neighbor peak.

These extractors should take the time shift already into account at the extraction step.

This is a bit more complicated, since time shifts might and will be non-integer. For this, we could imagine shifting the waveforms by the closest integer before extracting global / neighbor peak and after that applying the remaining correction after the extraction.

watsonjj, 2020-10-14:
>Ah, great. The question is then if a correction to be applied during DL0 → DL1 is correct in the DL1 container.

Its in the `DL1CameraCalibrationContainer`, not the `DL1CameraContainer`

kosack, 2020-10-14:
Can you explain better why this shift should change *per event*? I woudl assume it is like spatial flat-fielding, in that you compute it using a dedicated flasher run (or interleaved flasher events), and then it should not change on a long-term timescale.  In that case it would be best put in *Monintoring* data.  If there is a per-event time shift, I would call that a different type of calibration and not mix it with time flatfielding.

maxnoe, 2020-10-14:
This changes due to low-level DRS4 properties which depend on the physical capacitors used to sample the event.

maxnoe, 2020-10-14:
>  If there is a per-event time shift, I would call that a different type of calibration and not mix it with time flatfielding.

In the end, we will always calculate a correction for each event and put that somewhere in the event.
This might be computed from slow changing monitoring data and then filled into the event or it might be truly event dependent.

So having this information at the event make sense anyways.

watsonjj, 2020-10-14:
> This is a bit more complicated, since time shifts might and will be non-integer. For this, we could imagine shifting the waveforms by the closest integer before extracting global / neighbor peak and after that applying the remaining correction after the extraction.

Based on the conversation earlier in this thread and in https://github.com/cta-observatory/ctapipe_io_lst/pull/44 I would vote for performing the integer shifts to the waveform before charge extraction (or during). This is likely all that is needed for Intensity Resolution requirements. And then sub-sample corrections are only applied to the extracted peak_time value.

watsonjj, 2020-10-14:
> So having this information at the event make sense anyways.

@kosack Just to remind you we agreed on this per-event calibration container for the charge calibration fields too (also in `DL1CameraCalibrationContainer`). The factors used may be interpolated from between two monitoring events based on the waveform event time.

moralejo, 2020-10-14:
> Can you explain better why this shift should change _per event_? I woudl assume it is like spatial flat-fielding, in that you compute it using a dedicated flasher run (or interleaved flasher [...]

As @maxnoe said, this is a DRS4 feature, there is a ~ns delay which depends on where in the ring buffer a pulse falls, hence it changes per event. Actually, for most pixels this change is correlated, so it would not affect relative time resolution, but for other pixels it is not, so a per-event correction seems unavoidable to reach the best possible resolution.

maxnoe, 2020-10-14:
@moralejo I think this is only correlated for pixels that share the them DRS4 chip. Otherwise, it will be randomly distributed in the ring.

maxnoe, 2020-10-14:
> there is a ~ns delay

Calling it a delay is maybe not good, since it can also be negative. And in the same event, it can be positive for some pixels and negative for others.

watsonjj, 2020-10-14:
So what do we think is better? 

1. Shifting the waveforms in a common way by integer samples (detached from the charge extraction, and done before it. 
2. Only accounting for the integer shift within the charge extractor.

I feel the 1st is less error prone and easier to test.

FrancaCassol, 2020-10-14:
Hi @watsonjj,

conceptually speaking, it seems to me that it would be better to add the time calibration to the calibrator and not to the extractor. That means to add step 1 inside the calibrator before charge extraction and  to complete the time calibration adding the  remaining float shift to the pulse time (together with the usual camera time flatfielding shift )

maxnoe, 2020-10-14:
Sounds good, will make a PR soon

moralejo, 2020-10-14:
> @moralejo I think this is only correlated for pixels that share the same DRS4 chip (so only two pixels per gain for LST). Otherwise, it will be randomly distributed in the ring.

The Domino waves are synchronized, so if the "feature" is shared, the time shift it is correlated. But it is not for a sizeable fraction of pixels, which may point to differences in this feature among chips.

maxnoe, 2020-10-14:
> The Domino waves are synchronized, so if the "feature" is shared, the time shift it is correlated. But it is not for a sizeable fraction of pixels, which may point to differences in this feature among chips.

In my understanding of DRS4, the synchronisation of multiple DRS4 chips means, that capacitor 0 is sampled at the same time for all chips but the inhomogenous sampling of the ring is a hardware property of each chip, that can look very different from chip to chip.

So I don't really expect a correlation here, other than that these time shift curves will look similar for groups of pixels. Some show a sine-like shift over capacitors, some a more sawtooth like structure.

moralejo, 2020-10-14:
> In my understanding of DRS4, the synchronisation of multiple DRS4 chips means, that capacitor 0 is sampled at the same time for all chips but the inhomogenous sampling of the ring is a hardware property of each chip, that can look very different from chip to chip.
> 
> So I don't really expect a correlation here, other than that these time shift curves will look similar for groups of pixels. Some show a sine-like shift over capacitors, some a more sawtooth like structure, etc.

Below is a plot from @pawel21 showing what I mean. This is from flatfield events (~80 pe pulses). 

[image]

All of these are RMS of times (for many flatfield events), one entry per pixel. Blue histogram is with no correction, so it includes the "trigger_time - laser_shot" jitter. Red is the RMS of times relative to the camera mean, so that you remove the mentioned jitter. You can see that about 70% of pixels already have rms between 0.3 and 0.7 ns. Finally, the green curve is after correcting for the DRS4 shifts, i.e. final time resolution. For 70% of pixels the result is only marginally better after correction (~0.1 ns reduction of the rms), while for the remaining 30% which had rms 0.7-1.7 ns, the correction has an obvious effect and after it they look perfectly ok. It is for that 30% that the correction is most relevant.

maxnoe, 2020-10-14:
That probably means that these 70% have a similar shape of the non-uniform sampling and the remaining have a different one.

This should be easily verifiable by looking at the time shift vs. capacitor curves of the pixels that react well to this correction as compared to those who don't

maxnoe, 2020-10-14:
Ah, and my favourite question: this is not RMS right? This is standard deviation.

moralejo, 2020-10-14:
> Ah, and my favourite question: this is not RMS right? This is standard deviation.

Indeed, std. dev.  This is a bad habit from those of us used to ROOT, which kind of consolidated that misnomer.

maxnoe, 2020-10-14:
It is really impressive what a deep impact that software had on the problems of particle physicists talking to statisticians.

maxnoe, 2020-10-26:
The corresponding PR is now merged. #1463

closed 2020-10-26 by maxnoe