cta-observatory/ctapipe #1397: DL1 calibration's relative_factor / absolute_factor are confusingly named, documented and applied

MaxNoe, 2020-07-22:
The documentation of these factors makes no mention that in fact, that the `absolute_factor` is used not as factor but as divisor.

they are applied as :

``
charge *= relative / absolute
```

But the documentation of these values never mentions that and I think the naming is very confusing.

Also, why do we have these two factors in the same step? Why are these different calibration constants?

watsonjj, 2020-07-22:
So this came from the equation I believe is used in HESS, but with the naming changed:
relative = flat field (camera average divided by pixel value, when illuminated with uniform light source)
absolute = gain (mV per p.e.)
The naming was changed in an attempt to make this more explicit, but I agree I wasn't very successful...

watsonjj, 2020-07-22:
I believe they are kept in separate values as they are updated on different timescales

maxnoe, 2020-10-26:
After implementing #1463 and discussing with @watsonjj, we had the following idea for a simplification:

The `DL1CameraCalibrationContainer` has just one field for this amplitude correction factor, e.g. `flatfielding`.

This would contain the total necesseray corrections for an individual event, which would make the code in `CameraCalibrator` simple (just `charge *= flatfielding`) and avoid these confusingly named and differently applied `relative` / `absolute` factor / divisor.

The complexity would then be shifted where it actually belongs: filling the correct value for an event from different monitoring data.

Application is simple, calculating the correct calibration value is hard. So the complexity should be where it is conceptually.

maxnoe, 2023-10-23:
Since in the CTAO data model, R1 data will already contain the absolute calibration and we will only need to apply a relative update computed from interleaved calibration events, we should really implement the above conclusion.

maxnoe, 2024-09-03:
> So I would disagree with this issue partially (I agree the naming and one being multiplicative and the other a divisor is bad):

Similar comment to my comment in the PR: this issue is about changing the field in the event data structure, not the monitoring data.

I think it is good to move the complexity of combining the different monitored quantities like gain and other relative pixel differences to the monitoring code that computes the correction for the current event and only have the result of that computation in the event data structure.