cta-observatory/ctapipe #1406: Requirements for a new config system

MaxNoe, 2020-08-06:
I start a new issue here to collect the actual requirements in the top post here so we can discuss them. 

Previous discussions can be found here #1197 #1148 

### Top  Level Requirements

- [ ] Support for config files in a reasonable format (e.g. toml, yaml, json)  
- [ ] CLI support
- [ ] Configuration of instances possible (e.g. cleaning1, cleaning2 at the same level of hierarchy with different settings)
- [ ] Exporting the full config for provenance
- [ ] Runtime configuration possible through normal code
- [ ] Configuration options should be settable by ids / types / other properties (e.g. telescope id / type / camera type)

### Config file formats

I nowadays lead towards toml. There are a lot of discussions out there, but the gist is:

* JSON  
  + Simple
  + In the standard lib
  - Hard to edit for humans
  - No comments
  - No support for special floats `Inf`, `-Inf`, `Nan`

* YAML
   + Nicer to edit for humans
   + Supports pretty much everything
   - VERY Complex and a security hazard by default (running arbitrary code on deserialization)
   - Not in the standard lib, PyYAML unmaintained for a long time (seems no maintained again) but still on YAML 1.1, alternative ruamel.yaml 

* TOML  
  + Human readable
  + Used now in python for pyproject.toml
  + Far less complex than yaml but should support everything we need.
  o good library support but not in the std library
  - not yet at 1.0, but coming very soon (1.0rc1 is out now)

### CLI support

It would be nice to generate the full CLI from the configuration system, like now is done with the tools.
However, some aspects are currently not optimal:

* No positional arguments

### Configuration of instances

This should simply be done by configuring by attributes not class names, e.g. when having something like this:

```python
class Foo(Configurable):
     value: int

class Bar(Configurable):
    foo1:  Foo
    foo2: Foo

```
The config should look like this:
```python
{'foo1': {'value': 1}, 'foo2': {'value': 2}}
```

### Runtime configuration possible through normal code

This means, taking the same example above, that it should also be possible to do
```
Bar(foo1=Foo(value=1), foo2=Foo(value=2))
```

Instead of passing a config object.

### Proposals

I have to yet found any library or combination of libraries that offers this, especially not the configure by id / type part.
So maybe we need to roll our own, but based on libraries to 
* load the files (toml / yaml / json)
* build the CLI (click, ...)

kosack, 2020-08-06:
as in #1197 pydantic was pretty nice and provided many of the features above, but there was still one blocking feature (allowing member functions and variables that are not configured). Doesn't seem to have been solved, though there was a PR open.  Either TOML or YAML sounds fine to me. Even if solved, it would still require some work on our part to get all the features we use, but at least it would be much cleaner/documented than traitlets.config. 

We could consider having 2 separate systems:
- the object configuration system, with config file support, but not command-line
- command-line config options that allow high-level args (click or something), and also a way to set low-level parameters if really needed (basically just a special command-line option like  `--set-option "X.Y.value:3"`, which would be split into a dict of `{'Y': {'Y': 'value':3}}}` and sent to the config system just like a config file). 

As a side note,  we already support JSON or *Python* config files. I agree JSON is not nice to edit, but actually the Python ones that traitlets provides are pretty nice - they allow you do to some complex config things like use envvars, or get the current user name, etc, and are fairly readable.

kosack, 2020-08-06:
For other reference, I had started at some point to explore different config systems in some Gists below (these were done a while ago, and are not fully functional, but may be useful for somebody):

* Pydantic:  https://gist.github.com/kosack/da8fe7ef6b50e555939e7e932df99adf
* from scratch: https://gist.github.com/kosack/8158a2748a458bbece74a010abb02d3c

MaxNoe, 2020-08-07:
This TOML library is nice because it maintains style and comments: https://github.com/sdispater/tomlkit