cta-observatory/ctapipe #1410: black pre-commit hook in documentation

nbiederbeck, 2020-08-27:
I stumbled across the pre-commit hook for black in #1345.

I could not find any other mention of this. I suggest we add this somewhere in the [Getting Started For Developers](https://cta-observatory.github.io/ctapipe/getting_started/index.html#getting-started-for-developers) guide. Either:

* as a last [`get-the-ctapipe-software` step](https://cta-observatory.github.io/ctapipe/getting_started/index.html#get-the-ctapipe-software); or 
* as a first [`developing-a-new-feature-or-code-change` step](https://cta-observatory.github.io/ctapipe/getting_started/index.html#developing-a-new-feature-or-code-change)

In the same run we can think of adapting the [coding style guide](https://cta-observatory.github.io/ctapipe/development/style-guide.html#coding-style) which suggests following PEP8, which is not *really* necessary with the pre-commit hook.

Thoughts?

maxnoe, 2020-10-26:
Yes please, could you do that?

nbiederbeck, 2020-10-26:
PR: #1488

maxnoe, 2020-10-26:
You can link PRs in the panel on the right side ;)

closed 2020-10-27 by kosack

HealthyPear, 2022-02-02:
I stumbled also upon this...

I was thinking, wouldn't be easier to use the hook directly into the CI?
It is explained [here](https://pre-commit.com/#github-actions-example), so that the developer doesn't even need to worry about setting this up?

kosack, 2022-02-02:
Sure, maybe open a PR?

commit 4d7daa6d (2020-10-27): DOC: Add black pre-commit hook, fix #1410. (#1488)