cta-observatory/ctapipe #1412: Saving interactive html containing Bokeh camera display?

moralejo, 2020-09-04:
https://github.com/cta-observatory/ctapipe/blob/e65311dd6dc49c1314ad82be0c5c907a1fb6cc96/ctapipe/visualization/bokeh.py#L18

I would like to produce, using the bokeh camera display, an interactive html file in which a given camera display shows different values according to the position of a slider (the values would be run-wise averages of various pixel quantities, and the slider would be used to select the run). It seems this requires to use JavaScript callbacks, i.e. https://docs.bokeh.org/en/latest/docs/user_guide/interaction/callbacks.html.  What I need is something like the first example there, in which the slider changes the power of a polynomial, but in my case the plot would be a camera display. I gave it a try, with no success, and I am completely lost. Is it possible to do this with the current ctapipe bokeh camera display?

MaxNoe, 2020-09-04:
Can you show the code you currently have?

moralejo, 2020-09-05:
I just tried to follow the example in the linked bokeh page above, and make a camera display in which I could scale the pixel contents by a factor via a scaler. Actually I know that what is wrong there (at least one thing): the 'image' I am modifying does not contain at that point the pixel values, but the rgb color codes, so it _cannot_ work. But I do not know how to do it, since the camera.cdsource does not have the pixel values inside, just the colors.

> from bokeh.layouts import column
> from bokeh.models import ColumnDataSource, CustomJS, Slider
> from bokeh.plotting import Figure, output_file, show
> 
> import numpy as np
> from ctapipe.instrument import CameraGeometry
> from ctapipe.coordinates import EngineeringCameraFrame
> from ctapipe.visualization.bokeh import CameraDisplay as BokehCameraDisplay
> 
> 
> geom = CameraGeometry.from_name('LSTCam')
> engineering_geom = geom.transform_to(EngineeringCameraFrame)
> 
> pad_width = 350
> pad_height = 370
> z0 = np.linspace(0, 1854, 1855)
> cam = BokehCameraDisplay(engineering_geom)
> cam.image = z0
> cam.add_colorbar()
> 
> output_file("js_on_change.html")
> 
> source = cam.cdsource
> plot = cam.fig
> 
> callback = CustomJS(args=dict(source=source), code="""
>     var data = source.data;
>     var slider_value = cb_obj.value
>     var image = data['image']
>     for (var i = 0; i < image.length; i++) {
>         image[i] = image[i] * slider_value
>     }
>     source.change.emit();
> """)
> 
> slider = Slider(start=0, end=1, value=1, step=1, title="factor")
> slider.js_on_change('value', callback)
> 
> layout = column(slider, plot)
> 
> show(layout)

MaxNoe, 2020-09-06:
Ok, this could be solved by the changes O had planned for the camera display.

I will open a PR next week.

kosack, 2020-09-14:
I guess you're trying to make a stand-along gui? You can also do it in a Jupyter notebook with the interact tools, which work well with Bokeh displays (using `push_notebook(handle)` to update). Then you have access to all info, and no need for JavaScript.  Of course then you have to use the ipywidgets slider, not the Bokeh ones, I think.

kosack, 2020-09-14:
Alternately, use a Bokeh Datasource, and make sure add the pixel values to it (as in my example), so then you have access to those values not just the colors.

MaxNoe, 2020-09-14:
The problem is that the CameraDisplay is converting values to colors directly and not through bokeh means for mapping data to colors.

kosack, 2020-09-14:
Ah, yes, that is solved in the example above - just use a DataSource and ColorMapper.  I think we can just replace some of the existing code with what I showed in #1247 (with hex pixels as well).

moralejo, 2020-09-14:
Thanks a lot, @kosack, indeed with the display version in #1247  I managed to make an interactive camera plot and save it as html - I think that is very convenient for easy display of e.g. data quality checks to be accessed at a web site. Would be great to have this new CameraDisplay in one of the coming releases.
Eventually, one might have a camera display with the "multiple values" option built-in, i.e. rather than providing a single image to it, we could provide an array of images, mapped to certain values to be selected interactively with a slider (e.g. run numbers).

maxnoe, 2022-09-23:
Implemented in the new bokeh camera display

closed 2022-09-23 by maxnoe