cta-observatory/ctapipe #1414: Support multiple input files or appending to the output file for the stage1 tool

MaxNoe, 2020-09-10:
During todays aswg meeting it came up that the prod5 proton production SimTel files are rather small and if processed one-by-one this would result in very many, very small DL1 files (~250.000).

It probably would be good to be able to run the stage1 on multiple simtel files and produce a single output or to be able to append to an existing output file to achieve the same by running the tool multiple times.

bregeon, 2020-09-10:
The reasons for this to happen is two fold:
1. jobs can run on farms usually from 12 to 24 hours, so that fixes the number of showers per job, hence the DL0 file size
2. DL1 file size is much smaller than DL0 file size, so you want some merge/reduce to happen in the number of files so that the analysis is then more efficient

Indeed, I think that part of the workflow will stay like that: either you have directly multiple input files and one output, or a way to append to a DL1 file or you have a merging tool.

bregeon, 2020-09-10:
To give one example, for Prod5, twe have 112000 DL0 files per site.

MaxNoe, 2020-10-05:
For now, a tool to merge the output of the DL1 stage seems sufficient. 

@Hckjs  is working on that

closed 2020-11-02 by maxnoe