cta-observatory/ctapipe #1416: Writing instrument config to ecsv broken

MaxNoe, 2020-09-14:
Using `ctapipe-dump-instrument --format ecsv` is broken on the current master:

```
ctapipe-dump-instrument --infile=gamma_20deg_180deg_run20___cta-prod3-demo-2147m-LaPalma-baseline.simtel.gz --format=ecsv

yaml.representer.RepresenterError: ('cannot represent an object', PosixPath('/home/maxnoe/Uni/CTA/ctapipe/../simtel/Data/sim_telarray/cta-prod3-lapalma/0.0deg/Data/gamma_20deg_180deg_run20___cta-prod3-demo-2147m-LaPalma-baseline.simtel.gz'))
INFO [DumpInstrumentTool] (tool/run): Output:
```

kosack, 2020-09-14:
I think the solution from here might work:
https://stackoverflow.com/questions/21695705/dump-an-python-object-as-yaml-file

Specifically, this: (or probably better yet to add a representer for `pathlib.Path` objects explicitly. )

```py3
import yaml

yaml.SafeDumper.yaml_representers[None] = lambda self, data: \
    yaml.representer.SafeRepresenter.represent_str(
        self,
        str(data),
    )
```

maxnoe, 2021-02-12:
Fixed in #1451

closed 2021-02-12 by maxnoe