cta-observatory/ctapipe #1421: Make output path of provenance.log configurable

MaxNoe, 2020-09-16:
This is currently hard coded to `provenance.log` in `cwd` which makes it impossible to run the stage1 tool in a read-only directory.

kosack, 2020-09-25:
probably there needs to be a much more general discussion of how we store outputs. Ideally this file would be "attached" somehow to all output files produced by a Tool.  Though since the output files contain an activity_id and a data_product_id, that may be sufficient  

At minimum I'd suggest we make the filename and the output of provenance configurable.  

Note that the provenance system will likely be overhauled soon (there have been several meetings, and ongoing prototype work  to support IVOA prov,LogProv, etc.)

Bultako, 2020-09-25:
Could we think of having this path and filename as a property of the Tool that can be modified? (with a default value `provenance.log`) In this case we take the risk of ending up with different provenance files for different tools, instead of a single one, which frankly I do not see as a problem now..

MaxNoe, 2020-09-25:
Yes, this would be the implementation

MaxNoe, 2020-09-25:
Also, since it will be needed by everything in cta, provenance should probably be handled its own, self contained library with minimal dependencies, so it can be used by every other package as dependency

closed 2020-10-08 by kosack