cta-observatory/ctapipe #1425: Remove any reliance on the guessed names / types for the analysis

MaxNoe, 2020-09-16:
No part of the actual analysis / file io should rely on these guessing keys.
These should only be used for convenience when human interaction is the target, e.g. for plot labels.

kosack, 2020-09-25:
I'd also add: it is ok to rely on a standardized `telescope_type_id` which is what we call "telescope_name" in ctapipe now, but given that there is no standard right now and they are not stored or defined in the SimTel file, it's prone to errors.  So either we just assign relative integers  for types so they are at least internally consistent in a single file (a bit less human readable and not so easy to compare between files), or we ask for standard identifiers for optics and cameras to be added to the SimTel output and also to real data (probably not so easy, and lacking a good definition as of now).

MaxNoe, 2020-09-25:
At the moment we don't have these standardized ids and asking for them in the future is no solution for the already existing and computationally expensive LST/MAGIC simulations where this problem occurs, see #1424.

So I think the only solution for now is using internally consistent indices like proposed in #1426

maxnoe, 2023-01-04:
This was fixed, at least for newer simulations using the metadata system.

closed 2023-01-04 by maxnoe