cta-observatory/ctapipe #1430: Store "Angle to observing position (all showers)" simtel histogram in stage1 tool

MaxNoe, 2020-09-24:
This histogram is needed for the normalization of full enclosure IRFs

kosack, 2020-09-25:
Can you be a bit more specific in the description? It's not obvious to me from the one sentence given.   Why is (r,phi) per event is not sufficient? What does this have to do with "full-enclosure" (meaning `theta_square_max == inf`) IRFs?

MaxNoe, 2020-09-25:
This is the viewcone distributionof all the simulated events. It's the hist with id 3.

This is needed to normalize the number of selected events to the total  number of simulated events in bins of the fov offset angle.

MaxNoe, 2020-09-25:
The part in quotes is the actual title of the histogram in the simtel files.

I am already working on a PR.

kosack, 2020-09-25:
Thanks, that is more clear.