cta-observatory/ctapipe #1435: OBS_ID / EVENT_ID is not unique for current simulations

MaxNoe, 2020-10-01:
This came up during pyirf developement.

For the current Prod3b and Prod5 simulations, the run numbers are only unique inside each monte carlo set (particle type, array, pointing, viewcone). So when e.g. merging multiple pointings, `OBS_ID`, `EVENT_ID` is no longer unique.

kosack, 2020-10-01:
that is very annoying, and somewhat breaks the definition of an ID.  We could artifically change the obs_id, e.g. by including the particle type or something, though that's a bit ugly.  E.g. obs_id=12345 → N0012345, where N is the particle ID. 

I don't really like hacking things like that, it would be much better to require the simulations have unique obs_ids.

kosack, 2020-10-01:
I guess this needs to be a requirement somewhere... I could add it to DPPS Level-B reqs that obs_ids generated by the Simulation Pipeline System shall be unique.

GernotMaier, 2020-10-01:
My point is that we should not rely on the internal numbering in CORSIKA. We don't do that for the event numbers (which are re-assigned in sim_telarray). What we need is a unique obsID / event ID at the DL0 level - which is the case already now if you simply take the 0deg / 180 deg into account in the current scheme (agree that this is not ideal).

MaxNoe, 2020-10-01:
> which is the case already now if you simply take the 0deg / 180 deg into account in the current scheme (agree that this is not ideal).

No, it's not the case and it's not clear. Because there is no way to define a unique obs id from that information alone.

GernotMaier, 2020-10-01:
Why not? production_runnumber_eventnumber_az is at this point a unique ID.

LukasNickel, 2020-10-01:
That is more than the pointing and the run_id though.

Besides the inconvenience, the point still stands, that this needs to be documented and used everywhere in order to identify events. Which is, as of now, not the case, as at least ctapipe does not take this into account.

kosack, 2020-10-01:
It's ok to generate a new OBS_ID from existing metadata when we read the simulation files, but as right now we do not have that implemented and what we write to the DL1 files is so far just the sim_tel run_id.   At the very least, I would like to see a recipe on how to generate this id, preferably in functional form, something like:

```py3
def generate_simulation_obs_id(
    simulation_run_id : int,  
    simulation_azimuth_bin: float,
    simulation_particle_id: int,
) -> int:
    # fill in code here 
    return unique_obs_id
```

LukasNickel, 2020-10-01:
> We then have to always record the inputs to this function as metadata in any output tables, not just the run_id/obs_id as now...

As long as the generative function is bijective, this should not be needed, or?

bregeon, 2020-10-01:
Reporting comments from Slack:
we should decide up to which point they have to be unique, right now it's by prod/site/particle/zenith/pointing.
As mentioned above, there is a limit on Corsika run number <999999, and prod/site/zenith is 112000 jobs already in Prod5, and we plan 3 zenith angles at least (20,40,60).
To be precised, Prod5 also already runs 2 levels of NSB, so we have each run in 2 levels of NSB from the same Corsika showers.
And we might want to run a brighter NSB for Paranal.
And from one Prod to another, the number of runs might double to have more efficient jobs throughput on the grid.

MaxNoe, 2020-10-01:
But if we'd push the limit from 6 to 9 digits, this would be enough for everything right?

MaxNoe, 2020-10-01:
I checked the CORSIKA code, it should be changes in three lines.

MaxNoe, 2020-10-01:
I wrote Heck and Pierog a mail asking to extend the run number to 9 or 10 digits.

maxnoe, 2020-10-26:
They will implement an option to extend the number of runs to 9 digits.

closed 2020-10-26 by maxnoe