cta-observatory/ctapipe #1436: Strange behavior in Tools when checking invalid arguments

Bultako, 2020-10-01:
I have found that Tools do not always capture bad or invalid arguments.
Some examples below.

```
MyTool --foo=True
WARNING [MyTool] (loader/load_config): Unrecognized alias: '--foo=True', it will probably have no effect.
INFO [MyTool] (tool/initialize): ctapipe version 0.8.0.post31+gite99e39c
```

```
MyTool --Foo=True
WARNING [MyTool] (loader/load_config): Unrecognized alias: '--Foo= True', it will probably have no effect.
CRITICAL [MyTool] (application/catch_config_error): Bad config encountered during initialization:
CRITICAL [MyTool] (application/catch_config_error): Invalid argument: '--Foo=1'
```

```
MyTool --foo.bar=True
INFO [MyTool] (tool/initialize): ctapipe version 0.8.0.post31+gite99e39c
```

```
MyTool --Foo.bar=True
INFO [MyTool] (tool/initialize): ctapipe version 0.8.0.post31+gite99e39c
```

I guess this could produce an unexpected behavior of the Tool if a misspelled argument is passed together with a config file having the right argument name, which will turn out sometimes in a silent execution of the Tool.

MaxNoe, 2020-10-01:
Yes, it seems non-top-level config options are just silently ignored.

I could have sworn I already opened this issue, but I could not find it. So maybe I forgot to open it.

kosack, 2020-10-07:
perhaps add this to #1406 (reqs for a new config system).

kosack, 2020-10-07:
also related to https://github.com/ipython/traitlets/issues/145