cta-observatory/ctapipe #1438: TwoPassWindowSum image extraction crash using ctapipe 0.9.0

HealthyPear, 2020-10-02:
**Example**

simtel file = gamma_20deg_180deg_run100___cta-prod3-demo-2147m-LaPalma-baseline.simtel.gz
EVENT # 2385
EVENT_ID # 2441402
TEL # 1

**Error**

```
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/ctapipe/calib/camera/calibrator.py", line 154, in __call__
    self._calibrate_dl1(event, telid)
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/ctapipe/calib/camera/calibrator.py", line 126, in _calibrate_dl1
    waveforms, telid=telid, selected_gain_channel=selected_gain_channel
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/ctapipe/image/extractor.py", line 1060, in __call__
    waveforms, telid, selected_gain_channel, charge1, pulse_time1, correction1
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/ctapipe/image/extractor.py", line 900, in _apply_second_pass
    timing = timing_parameters(camera_geometry, image_2, pulse_time_1stpass, hillas)
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/ctapipe/image/timing.py", line 74, in timing_parameters
    beta, error = lts_linear_regression(x=longi, y=peak_time, samples=5)
  File "/Users/michele/Applications/miniconda3/envs/protopipe/lib/python3.7/site-packages/numba/np/linalg.py", line 825, in _inv_err_handler
    "Matrix is singular to machine precision.")
numpy.linalg.LinAlgError: Matrix is singular to machine precision.
```

kosack, 2020-10-02:
Can you check to see what these bad images look like? (I suspect few pixels survive so the time slope is not easy to define)

The solution will be to either:
1. catch this exception and just return NaN for the slope for those images (if they are really not measureable)
2. use SVD to invert the matrix (but I just tested that an it is 8x slower than `np.linalg.inv`, even when numba jitted)

So let's hope that case 1 is ok.

MaxNoe, 2020-10-02:
Fix in #1439 

I suspect that this happens in the loop over the pairs of points for the initial fit. So probably, the result of lts_fit should still be non-nan for the data.

But a test dataset would be great for that.

MaxNoe, 2020-10-02:
Okay, I added test cases for this in #1439.

I think with this, it should be really, really rare to get nan back from the fit, but it might happen and should be handled gracefully by the pipeline.

kosack, 2020-10-12:
This is fixed now.

closed 2020-10-12 by kosack