cta-observatory/ctapipe #1440: Uncertainties on Hillas Parameters.

andres-baquero, 2020-10-02:
I want to comment about the possibility to implement the computation of the uncertainties of the Hillas Parameters on ctapipe. 

Some time ago we presented on an ASWG call an analysis using the Hillas’s uncertainties to train Random Forest models, here's the link [Hillas's uncertainties](https://indico.cta-observatory.org/event/2348/contributions/20708/attachments/15797/20243/Hillas_Uncertainties.pdf).

We only include the uncertainties of Length and Width that results from the poissonian fluctuations on the number of photons per pixel.

It would require introducing new variables for the uncertainties in dl1.

We would like to make it available at least as an optional feature.  What do you think would be the best way to implement it?

MaxNoe, 2020-10-02:
How expensive are these computations?

If the answer is "not very", I'd say just go ahead and add the computations to the standard `hillas_parameters` functions and the corresponding fields to the container.

kosack, 2020-10-05:
I agree with @MaxNoe: my main question would be: are the computations for the uncertainties purely analytic, or do they require trained models/lookups (meaning an extra analysis step to prepare to generate them)?  If only analytic, then go ahead and add them.  If not, they should be a separate container and function, as they would imply they can't be filled without additional computations.

andres-baquero, 2020-10-05:
Thank you @MaxNoe and @kosack for your comments.

Yes, the computations are purely analytic.

About how expensive these calculations are, I have to adapt my code to the latest version of the hillas_parameters () function in ctapipe to have an answer on this. So I will make this adaptation and inform you.

MaxNoe, 2020-10-05:
You can use this to check the impact on the performance:

```python
❯ cat perf/time_hillas.py
from ctapipe.image.tests.test_hillas import create_sample_image
from ctapipe.image.hillas import hillas_parameters
import numpy as np
from timeit import repeat

geom, image, clean_mask = create_sample_image("30d")

code = '''
hillas = hillas_parameters(geom[clean_mask], image[clean_mask])
'''
exec(code)

N = 1000
times = np.array(repeat(code, number=N, repeat=10, globals=globals()))
print(f'{np.mean(times / N) * 1e6:.2f} ± {np.std(times / N) * 1e6:.2f} µs')
```

andres-baquero, 2020-10-05:
Using the @MaxNoe script to check the impact of the computation of Hillas uncertainties in performance I got:

For hillas_parameters() without uncertainties = 607.19 ± 1.64 µs

For hillas_parameters() with uncertainties =741.83 ± 2.07 µs

maxnoe, 2020-10-05:
Ok, then I would say go ahead and open a pull request modifying the `hillas_parameters` method and the `HillasParameters` container

maxnoe, 2020-10-26:
Fixed by merging #1443

closed 2020-10-26 by maxnoe