cta-observatory/ctapipe #1441: twopass extractor charge resolution broken in v0.9.0

kosack, 2020-10-05:
@HealthyPear found that the Charge Resolution benchmark no longer passes, see below.  

Seems to be a bug introduced in either #1356 or possibly, but less likely, #1370 

For ctapipe 0.8.0:
[image]

For ctapipe 0.9.0:
[image]

kosack, 2020-10-05:
Note the problem occurs in the first pass of the TwoPass method, so is likely something simple and not related to the time fitting (which hasn't run at this point in the benchmark)

HealthyPear, 2020-10-05:
also, nevermind the requirement, is not in the same units...I have to remember to disable it for the time being...

kosack, 2020-10-05:
I think the problem is just in the definition of the windows and shifts. With the current version, there was a misunderstanding on the definition and it should miss the peak in many cases - working on a fix.

closed 2020-10-12 by kosack