cta-observatory/ctapipe #1448: naming scheme of tools in ctapipe

kosack, 2020-10-08:
Right now, the tool names are a bit of a mess.  I suggest defining a scheme for existing and future tools.  

Some terminology (not standard, just to avoid confusion below):
* merge = combine files vertically (e.g. concatenating together similar data for more stats)
* join = combine files horizontally (e.g. a DL1 file containing tel001 with DL1 file containing tel002)

We need at least these basics (and likely more once we include calibration, etc):

* stage 1 of pipeline (Rx/DL0 → DL1)
    * generate DL1/Event data
    * train algorithms needed for stage 1
    * merge DL1 files
* stage 2 of pipeline (DL1  → DL2)
    * join  DL1 files from  multiple telescopes
    * generate DL2 event lists
    * merge stage2 outputs
* perform stage 3 (DL1 → DL2) 
   * train algorithms needed for stage3
   * generate DL3  event list 
   * generate DL3 IRF
    
What is a good naming scheme? I tried to avoid the data level too much, since you might use the same tool to do multiple data levels.  The original idea was to make hierarchies of these using sub-Tool functionality, that could be called on a single command-line, similar to jupyter):

`ctapipe stage1 process `
`ctapipe stage1 train `
`ctapipe stage1 merge `
`ctapipe stage2 process `
`ctapipe stage2 merge `
`ctapipe stage3 irf `

but as in #1447 , we could have a single merge tool `ctapipe merge` that does both merging and joining of data at all data levels DL1+.  What are other suggestions for a nice consistent naming scheme? It basically should represent a high-level API, and note that these can also be used both as command-line tools and in Jupyter Notebooks.

maxnoe, 2020-10-08:
Merging is already needed before stage2, as you will most probably need to train the models on multiple simulation runs

kosack, 2020-10-08:
yes, merging is needed everywhere. Those are not meant to be in order (you likely need to merge before training, etc).

kosack, 2020-10-08:
a `ctapipe validate` tool could also be useful to ensure the output data is in a particular format (not science validation, but schema validation), and maybe a `ctapipe metadata extract` tool to extract metadata, etc.