cta-observatory/ctapipe #1449: Do not rely on Table serialize_meta for hdf5 files

maxnoe, 2020-10-09:
We should have a well-defined way of storing metadata like units in hdf5 files.

Right no it is a mixture of adding `*_UNIT` attributes in the HDFwriter and using astorpy's additional yaml based table (e.g. when writing the subarray info).

We should use one unified approach, preferably one that does not rely on astropy internals.

kosack, 2020-10-09:
I liked using the astropy system because I can re-use it in tools like ctapipe-dump-instrument to write to ECSV or FITS files that can still be read back in by "pure" astropy with no ctapipe code at all. But that just requires that the Instrument classes either store info internally in astropy Tables, or provide a to_table() function.  It kept the IO simple.  

For the HDF5 files, perhaps that is not necessary, and we could define a translation later like in #1427, while still keeping the ability to read/write directly to astropy Tables. 

One thing: when I originally implemented this, astropy (v2 It think) stored all these headers and metadata in standard HDF5 attributes. For some reason they deprecated that functionality, and used a YAML table attached, which I find pretty ugly.

kosack, 2020-10-09:
Really, I wish that HDF5 just had a nice standard for units. That would just solve it all, and you wouldn't need ctapipe-specific code to read the data.

maxnoe, 2020-10-09:
For most parts, we can keep this approach but with using `serialize_meta = False`. 
This will save all custom meta information if possible (fits headers, hdf5 attributes), it will even store units for fits, as there is a well-defined standard on how to do this. But it won't store units for hdf5 automatically, since hdf5 has no such standard.

But it could be as simple as:

```
for col in table.columns:
    if col.unit is not None:
        table.meta[col.name + "_UNIT"] = table.unit.to_string('vounit')
table.write(path, serialize_meta=False)
```

maxnoe, 2020-10-09:
I really don't like the second dataset, which was introduced in astropy 4 due to size constraints of the hdf5 attributes, but these shouldn't matter for us for storing the meta information cta requires.

commit 29237609 (2022-03-29): Merge pull request #1817 from cta-observatory/write_table

Add write_table and use it in SubarrayDescription.to_hdf, fixes #1449