cta-observatory/ctapipe #1456: Unit of peak_time

maxnoe, 2020-10-14:
The peak_time container is currently not defined with a unit.

The extractors convert them to ns.

Shouldn't we add this unit to the container / make it a quantity?

watsonjj, 2020-10-14:
> Shouldn't we add this unit to the container / make it a quantity?

I've generally been against attaching units to the image related values (#684 #1044). But the `peak_time` description should be updated to indicate its units.

> And related: should be also require any timeshifts be in ns instead of in adc sample width?

I think this might complicate the process, as the shifts are generally extracted in sample units, and need to be in sample units for any corrections involving the waveform (before peak_time extraction)...but of course this depends on what we choose to implement in ctapipe for the timing corrections (#1396).

maxnoe, 2020-10-14:
I agree that it might make it simpler to use samples for the corrections and the conversion is anyway simple, so we just have to decide.

My feeling is, for consistency, it would be better to have peak_time and time_shift in the same units and convert to samples where needed.

kosack, 2020-10-14:
I think I prefer having it in `ns` and then it's clear that to go to samples, you have to convert back to samples.  Especially if in the future we do some upsampling of the waveforms, this will avoid confusion. Nanoseconds is then a physical unit, and is common for all telescopes, which is nice (same reason we go to PE, and not digital counts).

However, if instead you want to stick with samples, then as elsewhere, it would be good to add the unit name to the Field name if units are not attached, e.g. rename it to `peak_time_samples`, so later it's clear when we do a conversion to `peak_time_ns`.

kosack, 2020-10-14:
Oh, one more argument for using nanoseconds: we may/will eventually do more pulse analysis than just the peak time in the future. For example extracting also the pulse width, rise time, etc.   In that case, doing that in samples is needlessly confusing.

watsonjj, 2020-10-14:
Ok, lets specify in the field descriptions that time_shift and peak_time is in units of time ("ns"). Makes sense from their names. But I still would prefer not attaching astropy units to them.

maxnoe, 2020-10-14:
Agreed.

moralejo, 2021-02-05:
> Ok, lets specify in the field descriptions that time_shift and peak_time are in units of time ("ns"). Makes sense from their names. But I still would prefer not attaching astropy units to them.

Why not for the peak times?

maxnoe, 2021-02-05:
I guess because of the computational cost that comes with astropy units. However, the cost is much less when not actually converting units but just accessing the values in the current unit.

Let me try actually measure the impact and if it is not big, I would be in favor of attaching the units.