cta-observatory/ctapipe #1459: many provenance-log-files get produced during tests

nbiederbeck, 2020-10-14:
While running the full `pytest` in the top level directory, 9 log-files get produced, which are "untracked files" for git:
```bash
$ git status --short
?? application.provenance.log
?? ctapipe-display-dl1.provenance.log
?? ctapipe-display-imagesum.provenance.log
?? ctapipe-display-integration.provenance.log
?? ctapipe-display-televents.provenance.log
?? ctapipe-dump-triggers.provenance.log
?? ctapipe-reconstruct-muons.provenance.log
?? ctapipe-stage1-process.provenance.log
```

maybe we can git-ignore them?

maxnoe, 2020-10-14:
Or pass a tmp dir for them now that these are configurable

maxnoe, 2020-10-14:
I would go for the tmp dir, to not have them at all in the repo, not even ignored.

kosack, 2020-10-14:
use the `tmpdir` fixture of pytest in the tests. https://docs.pytest.org/en/stable/tmpdir.html

nbiederbeck, 2020-10-14:
There seems to be some problems:
* the `provenance_log` trait is missing a `.tag(config=True)`
* it's missing an alias, eg. `self.aliases['provenance-log'] = 'Tool.provenance_log'` (?)
* after adding these two things, I get two log-files, the default and the one I provided, both having exact the same content (during the tests, not when running the tools myself)

maxnoe, 2020-10-14:
> the provenance_log trait is missing a .tag(config=True)

oh man, not again...

maxnoe, 2020-10-14:
It took me longer than I care to admit, but here it is, a script that scans all ctapipe classes for traits without config=True:

```python
import ctapipe
import pkgutil
import importlib

from ctapipe.core import Component, Tool

ignore_traits = {
    'config', 'parent', 'argv', 'log', 'name', 'subcommands', 'version', 'subapp', 'aliases',
    '_loaded_config_files', 'cli_config', 'description', 'examples', 'extra_args',
    'flags', 'keyvalue_description', 'option_description',
    'raise_config_file_errors', 'subcommand_description', 'classes'
}

def find_all_traitlets(module):
    module_name = module.__name__

    for submodule_info in pkgutil.iter_modules(module.__path__):
        submodule = importlib.import_module(module_name + '.' + submodule_info.name)

        if submodule_info.ispkg:
            find_all_traitlets(submodule)
        else:
            for obj_name in dir(submodule):
                obj = getattr(submodule, obj_name)

                try:
                    has_traits = issubclass(obj, (Component, Tool))
                except:
                    has_traits = False

                if has_traits:
                    for traitname, trait in obj.class_traits().items():
                        if not trait.metadata.get('config', False) and traitname not in ignore_traits:
                            print(f'{f"{obj.__module__}.{obj.__name__}":<80} {traitname}')

print('Traits without config=True')
find_all_traitlets(ctapipe)
```

Should we use that on travis?

nbiederbeck, 2020-10-15:
> oh man, not again...
> Should we use that on travis?

I would say yes, since you did the work now and this seems to be a common problem.

Does it exit with an error status? Otherwise we won't notice on travis.

maxnoe, 2020-10-15:
We can make it raise an error if it finds anything

maxnoe, 2020-10-15:
I will also get rid of the duplication

maxnoe, 2021-02-19:
Fixed in by running test tools in tmp dirs

closed 2021-02-19 by maxnoe