cta-observatory/ctapipe #1460: Peak time extraction is biased if there is a residual pedestal

maxnoe, 2020-10-14:
The `CameraCalibrator` only applies the `DL1CameraCalibrationContainer.pedestal_offset` to the integrated charge.

However, this residual pedestal needs also to be taken into account for the peak time extraction, which is biased (at least using our current weighted mean approach) if there is a residual pedestal.

I talked with @watsonjj and we came to the conclusion would be able to cure this:

* change the definition of the `pedestal_offset` to be the pedestal per sample in the waveform instead of a correction for the integral
* Subtract this pedestal from the waveform before doing the extraction

maxnoe, 2020-10-14:
This can be easily checked in the current camera calibrator tests which has a residual pedestal but does not test the correct peak time.

watsonjj, 2020-10-14:
This leads to the topic of what order should the calibration be applied in when extracting the shower images. Currently the order is:
1. Extract the charge and time
2. Subtract the integrated pedestal from the charge
3. Scale the charge by the flat-field coefficients to correct for non-uniformities
4. Convert the charge units into photoelectrons

The proposal here is to the subtract a pedestal first, before extracting the charge. Which (as @maxnoe indicates) is necessary for a correct result for the peak_time from using the weighted mean approach. This requires a different definition of the pedestal coefficient.

My question is what is the correct order we want to be applying calibrations? What does everyone expect?

maxnoe, 2020-10-14:
I think most extraction algorithms will expect a baseline of 0 or will be dramatically simpler if the baseline is 0.

So at least applying the pedestal before extraction seems like a must.

kosack, 2020-10-14:
To make sure I understand: I assume this means that the "pedestal_offset" in the MCs is really just the electronic (dark) pedestal and doesn't include the NSB noise?

watsonjj, 2020-10-14:
The pedestal subtracted in the CameraCalibrator is presumed to be the onsky pedestal (includes NSB noise). Dark pedestal is subtracted in the R1 calibration.

kosack, 2020-10-14:
So where does it get the Sky Pedestal currently? Or is that the problem, that it doesn't exist?

maxnoe, 2020-10-14:
I don't think it exists anywhere at the moment, but there is a unit test where it is non-zero which shows that the peak time extraction is biased for non-zero pedestal.

I uploaded the failing test here: #1461

maxnoe, 2020-10-14:
The test passes if the pedestal is zero or when the pedestal is subtracted from the waveform.

kosack, 2020-10-14:
Ok good, yes this is something I long suspected was a problem, since our charge bias was so large after extraction, and the optimized picture-boundary thresholds were different from MARS.  This will fix both of those problems.  

Is the level of NSB something we can read from the simulations, or do we have to compute the pedestals as we would in real data?  AS a first step just using what is in the sims would be the simplest, and later do the real computation on the fly, or read it from a Monitoring file.

watsonjj, 2020-10-14:
Does MARS then obtain a sky pedestal from the simulation files?

kosack, 2020-10-14:
No idea, I guess it just computes it from either past events, or the waveforms.

watsonjj, 2020-10-14:
Just because we are missing a sky pedestal subtraction at the moment in ctapipe because we don't fill the pedestal_offset values, does not necessarily mean that the current implementation is wrong. I believe the currently implemented approach is the way HESS does is. I would like to hear what approach others believe to be the correct one:
1. Offset the waveform so the average (when there is no signal) is zero before charge extraction
2. Calculate the integration result from waveforms with no signal, and subtract this value from the charge we measure when observing signals

What does MARS do? @moralejo @HealthyPear

maxnoe, 2020-10-14:
>  I believe the currently implemented approach is the way HESS does is

How does HESS then extract arrival time?

maxnoe, 2020-10-14:
FACT does 1

kosack, 2020-10-14:
I think HESS does 2, since we didn't have waveforms until recently, but I do like 1 better. Since we no longer have ASTRICam, all cameras give waveforms as well.

FrancaCassol, 2020-10-14:
Just to be sure to understand, are you speaking now for the level B (offline) calibration, ins't it?

maxnoe, 2020-10-14:
> but they don't exist in most simulations we have now)

I think they do. Not the interleaved pedestals, but the averages. In the CameraMonitoring objects

maxnoe, 2020-10-15:
Yes, the pedestal is read from `array_event['camera_monitorings'][tel_id]['pedestal']` but is only stored in the mc container, not the calibration container.

moralejo, 2020-10-15:
Hi,
in the MARS analysis of CTA MC (in its 'Chimp' converter-calibrator part), the pedestal baseline is subtracted before integration, following sim_telarray's analysis, the sample-wise value being 
`moni->pedestal[gain][ipix]/moni->num_ped_slices`

As for the 'sky' or electronic pedestal, at least for LST there is no difference, the baseline is the same (only noisier in the sky case).

I was surprised to learn of this problem with pulse time calculation, because I remember that at some point @watsonjj had solved a bug in it: negative sample values were being considered, which obviously makes no sense in such a weighted mean. How come did we have  negative sample values?  For pulse time calculation in the MARS analysis we use the weighting method, ignoring negative sample values.

kosack, 2020-10-15:
Ok, so if we put the SimTelArray Camera Monitoring pedestal into the same container that @FrancaCassol uses for the latest pedestal for real data in SimTelEventSource, then that should make things work similarly for  both,  even if the simulations don't have interleaved pedestals.  In the future we can make the pedestal computation algorithm choosable.

watsonjj, 2020-10-15:
> mean digital signal derived

This sounds ambiguous considering the discussion above on the two different approaches

watsonjj, 2020-10-15:
> Ok, so if we put the SimTelArray Camera Monitoring pedestal into the same container that @FrancaCassol uses for the latest pedestal for real data in SimTelEventSource, then that should make things [...]

Yes that would fill the monitoring containter. We still would need something to fill the `event.calibration.tel[telid].dl1.pedestal_offset` field if we want a pedestal subtraction to be applied when extracting DL1.

But I think this has made the preference clear that we want to use approach 1 for pedestal subtraction, which would also fix the problem in the tests that @maxnoe originally reported in this issue. So we proceed as you had planned @maxnoe.

FrancaCassol, 2020-10-15:
Hi @kosack 

> Ok, so if we put the SimTelArray Camera Monitoring pedestal into the same container that @FrancaCassol uses for the latest pedestal for real data in SimTelEventSource, then that should make things [...]

I think it makes sense. I already use those containers for simulated calibration and as you say one day we want MC and data  interchangeable    

> 
> Should we also rename some Fields to be clear which type of pedestal each is? e.g. `sky_pedestal`
> 
It is not so clear what we gain specifying the origin at this level. Actually here the only pertinent pedestals are the sky_pedestals, probably the other types will never go through this level of calibration code...

maxnoe, 2020-10-15:
And they could even be from multiple sources. The only thing import is that it contains the correct remaining pedestal, independent of source.

Which sources contribute is a problem for filling this value, not for applying it.

maxnoe, 2020-10-15:
We already subtract this pedestal in the `apply_simtel_r1_correction`. So the remaining pedestal should be zero.

maxnoe, 2020-10-15:
I implemented the changes discussed here in #1461, please review

commit f94fe682 (2020-10-15): Correct pedestal before extracting the pulse, fixes #1460 (#1461)

* Add failing test for peak time to show bias in extractoin

* Use correct pedestal width (was correct before by accident)

* Apply pedestal corection before peak extraction

This changes the definition of the pedestal value stored
in the DL1CameraCalibrationContainer to be the pedestal per sample
in the waveform, not the pedestal for the integral.

* Update help for pedestal

* Remove debug print

* Use more readable alternative to transposing

* Remove unused import

* Add missing import

* Adapt tests in simteleventsource to [... cut]
closed 2020-10-15 by kosack