cta-observatory/ctapipe #1480: Dtypes of some image parameters not fixed for some cases

maxnoe, 2020-10-20:
One file:

```
{('intensity_max', '<f8'),
 ('intensity_mean', '<f8'),
 ('intensity_min', '<f8'),
 ('intensity_std', '<f8'),
 ('leakage_intensity_width_1', '<f8'),
 ('leakage_intensity_width_2', '<f8'),
 ('peak_time_max', '<f8'),
 ('peak_time_mean', '<f8'),
 ('peak_time_min', '<f8'),
 ('peak_time_std', '<f8')}
```

Another one:
```
In [6]: b - a                                                                                                                                                                                                                                                                               
Out[6]: 
{('intensity_max', '<f4'),
 ('intensity_mean', '<f4'),
 ('intensity_min', '<f4'),
 ('intensity_std', '<f4'),
 ('leakage_intensity_width_1', '<f4'),
 ('leakage_intensity_width_2', '<f4'),
 ('peak_time_max', '<f4'),
 ('peak_time_mean', '<f4'),
 ('peak_time_min', '<f4'),
 ('peak_time_std', '<f4')}
```

I have the fear, that this is because `np.nan` is a float64, but the calculated image parameters might be float64.

maxnoe, 2020-10-20:
Checking 200 La Palma prod3 files, I get this distribution of dtypes in the telescope image parameter table for these columns:

```
In [16]: Counter(dtypes)                                                                                                                      
Out[16]: Counter({dtype('float32'): 2926, dtype('float64'): 878})
```

kosack, 2020-10-20:
specifying a dtype everywhere is a good idea, even for scalars. Originally the definition of "Field" was perhaps too lightweight, in that it didn't require the developer to specify everything and just inferred things from the values.  Moving toward being more strict will avoid these problems.

kosack, 2020-10-20:
> Checking 200 La Palma prod3 files, I get this distribution of dtypes in the telescope image parameter table for these columns:

Did this have to do with which machine did the analysis?

maxnoe, 2020-10-20:
All on the same machine. So I would think the float64 nan is the culprit

maxnoe, 2020-10-21:
For now fixed in the merge tool by casting to the dtype of the first file.

LukasNickel, 2020-10-21:
My monologue a few weeks ago: #1411

maxnoe, 2020-10-21:
closed as duplicate of #1411

closed 2020-10-21 by maxnoe