cta-observatory/ctapipe #1482: HDF5TableReader column transforms failing for enums

LukasNickel, 2020-10-21:
This occured after merging the current master into #1357.
When I now try to iterate events generated by the DL1EventSource, I get an error stating:
>TypeError("'numpy.str_' object is not callable")

This happens when calling this function:
 https://github.com/cta-observatory/ctapipe/blob/225d5eed16e1b3733252941f0849cf246652f4c4/ctapipe/io/hdf5tableio.py#L391

When the transform function is assigned to the column, the attr / colname are matching the expected (`event_type_ENUM`).
When the function is called, attr is now `time_DESC`, which does not point to an enum, but a string in the attributes. This is the last value from the loop. It seems reasonable to assume, that the value of `attr` is not fixed inside the function, but why did this work before?

As far as I can tell, that code has not been touched in a while and my python version is also fixed, so I really wonder why this is failing now all of a sudden.
I will try to generate a MWE tomorrow to investigate.

kosack, 2020-10-21:
the code to write column descriptions as `*_DESC` attributes was added recently - could there be an unintended interaction? The enum code should look only for the `*_ENUM` attributes, though

maxnoe, 2020-10-21:
The full traceback would be helpful

maxnoe, 2020-10-21:
Shouldn't the DL1EventSource only call it the other way around? `enum_to_int` is the the transform for writing, it should be `int_to_enum` for reading back?

LukasNickel, 2020-10-21:
> the code to write column descriptions as `*_DESC` attributes was added recently - could there be an unintended interaction? The enum code should look only for the `*_ENUM` attributes, though

The loop goes over all  attributes though. If the ENUM entry was the last one before this change, then that would explain this.

kosack, 2020-10-21:
> Do we have a notion of inverse transforms for the reader?

Yes

LukasNickel, 2020-10-21:
```
    def test_max_events(dl1_file):
        max_events = 5
        with DL1EventSource(input_url=dl1_file, max_events=max_events) as source:
            count = 0
            assert source.max_events == max_events  # stop iterating after max_events
            assert len(source) == 20  # total events in file
>           for _ in source:

ctapipe/io/tests/test_dl1eventsource.py:51:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
ctapipe/io/eventsource.py:255: in __iter__
    for event in self._generator():
ctapipe/io/dl1eventsource.py:156: in _generator
    yield from self._generate_events()
ctapipe/io/dl1eventsource.py:264: in _generate_events
    for counter, array_event in enumerate(events):
ctapipe/io/hdf5tableio.py:506: in read
    container[fieldname] = self._apply_col_transform(
ctapipe/io/tableio.py:167: in _apply_col_transform
    value = tr(value)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

int_val = 32

    def transform_int_to_enum(int_val):
        """transform integer 'code' into enum instance"""
        enum_class = tab.attrs[attr]
>       return enum_class(int_val)
E       TypeError: 'numpy.str_' object is not callable

ctapipe/io/hdf5tableio.py:402: TypeError
```

kosack, 2020-10-21:
However, i think it's just a misnamed function: the code does indeed transform int to enum

kosack, 2020-10-21:
Ah... so somewhere the "serialized" Enum (not idea how that even works) is being turned into a string in the headers. Is that a change in how pytables serializes objects?  To be honest, I didn't even know it could serialize objects - that was added by Dominik, I think, so perhaps he knows better?

LukasNickel, 2020-10-21:
> However, i think it's just a misnamed function: the code does indeed transform int to enum

But the function name is `transform_int_to_enum`, thats exactly what it does, no?

kosack, 2020-10-21:
Can you run this in the debugger and see what `tab.attrs[attr]` is? or just look in vitables perhaps?

maxnoe, 2020-10-21:
Wait, how did this ever work? Did pytables automatically pickle and unpickle the class? That's the only thing how I can imaging this working.

LukasNickel, 2020-10-21:
>  (not idea how that even works) i

> Wait, how did this ever work? 

Thats excactly me for the last hour

maxnoe, 2020-10-21:
It assigns the class here:

```
                    meta[f"{col_name}_ENUM"] = value.__class__
```

so it has to be pickle.

LukasNickel, 2020-10-21:
Attributes:

>   [CLASS := 'TABLE',
    CTAPIPE_VERSION := '0.9.1.post56+git83c5ced',
    FIELD_0_FILL := 0,
    FIELD_0_NAME := 'obs_id',
    FIELD_1_FILL := 0,
    FIELD_1_NAME := 'event_id',
    FIELD_2_FILL := 0.0,
    FIELD_2_NAME := 'time',
    FIELD_3_FILL := False,
    FIELD_3_NAME := 'tels_with_trigger',
    FIELD_4_FILL := 0,
    FIELD_4_NAME := 'event_type',
    NROWS := 20,
    TITLE := 'Storage of EventIndexContainer,TriggerContainer',
    VERSION := '2.7',
    event_id_DESC := 'event identifier',
    event_type_DESC := 'Event type',
    event_type_ENUM := <enum 'EventType'>,
    obs_id_DESC := 'observation identifier',
    tels_with_trigger_DESC := 'list of telescope ids with data',
    time_DESC := 'central average time stamp']

The `_f_list()` reduces this to:
>['CTAPIPE_VERSION',
 'event_id_DESC',
 'event_type_DESC',
 'event_type_ENUM',
 'obs_id_DESC',
 'tels_with_trigger_DESC',
 'time_DESC']

kosack, 2020-10-21:
if you do: 
```py3
o = some_object
x = 3
def internal_func(x):
    return o(x)
```
But then later change o, I think this would fail.  So it might have only worked if the ENUM attribute was the last one

LukasNickel, 2020-10-21:
> It looks a bit suspicious that it uses a closure instead of using functools.partial. Should check that the attrs object is not a reference that is changing - I suspect it might be.

Yeah, thats what I am thinking too. Which brought me to the question why it worked before.

maxnoe, 2020-10-21:
@kosack 

that's definetely it.

kosack, 2020-10-21:
Only by chance, that the ENUM was last in the list.  For other attributes I use functools.partial to ensure the associated data comes along with the transform function.

kosack, 2020-10-21:
But I'm actually still a bit surprised that the serialization worked at all!  I woudl have written custom attributes that are parsible, like `col_ENUM = "{'A':1, 'B':2'}"`

maxnoe, 2020-10-21:
```
In [11]: type(f.root.dl1.event.subarray.trigger._v_attrs['event_type_ENUM'])
Out[11]: enum.EnumMeta
```

Definetlely using pickle

maxnoe, 2020-10-21:
```
> In [16]: enum
Out[16]: b'cctapipe.containers\nEventType\np0\n.'
```

kosack, 2020-10-21:
[image]

Yup, and pytables even calls it a "Python" type. Didn't know it could do that.

kosack, 2020-10-21:
Probably not a safe thing to do.

maxnoe, 2020-10-21:
Now, we should change that.

It does not work if we change an enum definition or the enum is not imported, right?

kosack, 2020-10-21:
I suggest we:

1. don't define the transform like that, do it like the others `tr = partial(expand_enum(enum_class=e))`
2. make an `enum_to_json` and `json_to_enum` converter for the attribute.

maxnoe, 2020-10-21:
propably we need to do something like this:

```
json.dumps({k: v.value for k, v in enum.__members__.items()})
```

maxnoe, 2020-10-21:
we should support the old version though for the moment.

kosack, 2020-10-21:
Should be able to support both: if the attr is a serialized Enum, use it correctly, and otherwise use a JSON string.

LukasNickel, 2020-10-21:
>     1. don't define the transform like that, do it like the others `tr = partial(expand_enum, enum_class=e)`
> 
>     2. make an `enum_to_json` and `json_to_enum` converter for the attribute.

I can do that tomorrow

maxnoe, 2020-10-21:
Ok, that was a deep dive into python internals. And I don't think there is a good solution.

Code:

```
def enum_to_json(enum):
    data = {'name': enum.__name__}
    data['members'] = {k: v.value for k, v in enum.__members__.items()}
    return json.dumps(data)

def enum_from_json(enum_json):
    '''
    From a json description of an IntEnum, as created by enum_to_json,
    recreate the enum type.

    This goes somewhat deep into python class definition internals.
    '''
    data = json.loads(enum_json)
    name = data['name']

    # enum uses it's own special dict type as class dict,
    # which can be obtained through __prepare__
    enum_dict = enum.EnumMeta.__prepare__(name, (enum.IntEnum, ))
    for k, v in data['members'].items():
        enum_dict[k] = v

    # no with the correct dict type, we can create our new class
    return enum.EnumMeta.__new__(
        enum.EnumMeta, name, (enum.IntEnum, ), enum_dict
    )

```

That works fine:

```

In [1]: from ctapipe.containers import EventType

In [2]: from ctapipe.io.hdf5tableio import enum_from_json, enum_to_json

In [3]: json = enum_to_json(EventType)

In [4]: enum = enum_from_json(json)

In [5]: enum
Out[5]: <enum 'EventType'>

```

Until:

```
In [6]: enum.SUBARRAY == EventType.SUBARRAY
Out[6]: False
```

Because it's not really the same enum:

```
In [8]: enum is EventType
Out[8]: False
```

But the depickled one is:
```
In [15]: enum = pickle.loads(h5py.File('./test_dl1.h5', 'r')['/dl1/event/subarray/trigger'].attrs['event_type_ENUM'])

In [16]: enum is EventType
Out[16]: True
```

So I'd say there are three alternatives for the table reader:

1. Keep relying on pickle
2. Have a dict of known enums and perform a name lookup
3. Don't do anything with enums, just store the int value. 

If we go for 3, the DL1 event source could instantiate the trigger enum.

kosack, 2020-10-23:
I don't like the idea of the data model relying on Python-specific implementations (it shoudl be readable in any language).  Perhaps we just do both? I.e. store two attirbutes:
1. `<col>_ENUM_PY` that contains a pickled Enum class 
2. `<col>_ENUM` that contains a JSON string that at least allows the user to understand what each integer value represents. 

Then 1 can be used in ctapipe, and 2 is universally useful.

maxnoe, 2020-10-23:
yes, this would be a good solution I think.

closed 2020-10-23 by maxnoe

kosack, 2020-10-23:
Just always using integers is maybe easiest however, and the user can convert between those and Enums. `EventType(4)` returns `EventType.ELECTRONIC_PEDESTAL`, so one can just do:
```python3
if EventType(event.trigger.event_type) == EventType.HARDWARE_STEREO:
    # do something

# or

if event.trigger.event_type == EventType.HARDWARE_STEREO.type:
   # do something
```

maxnoe, 2020-10-23:
This is what is done now in the table reader and I think it's nice to have for convenience.

commit f00395f7 (2020-10-21): Fix enum deserialization in HDF5TableReader, fixes #1482
commit 010e77a5 (2020-10-23): Fix enum deserialization in HDF table writer (#1483)

* Add more fields to enum test

* Fix enum deserialization in HDF5TableReader, fixes #1482