cta-observatory/ctapipe #1484: Store event-wise pointing at least for MC

maxnoe, 2020-10-22:
I think we should *optionally* enable storing `dl1/event/telescope/pointing` with `TelEventIndex` at least for simulations since
the timestamp based approach will not work for mcs since they are produced in parallel.
Alternatively, we should atleast include an `obs_index` in the monitoring table.

This will create problems when merging files that do not only contain a single pointing position.

kosack, 2020-10-23:
what is `obs_index`? Is that `obs_id`? That would perhaps be better to store in the monitoring table along with the time, since then either could be used to link the dataset (in case the time is meaningless).

maxnoe, 2020-10-23:
yes, sorry.

kosack, 2020-11-05:
yes, at least adding an obs_id I think is useful.  Eventwise pointing could be done too, but it's really overkill to have the pointing position per event when it changes on a much slower rate (once per second + interpolation to the event time is usually enough).  So far simulations only have fixed alt/az - we will have to consider something better when and if we do ra/dec simulations.   I had always assumed we'd just have a function that just provides event_wise pointing based on the monitoring data, like a `PointingInterpolator`. For MCs, this could just use the obs_id if we add it, and for observations, it would be a proper interpolation of the actual pointing monitoring data.