cta-observatory/ctapipe #1493: where to store MC DL1 information in event structure

kosack, 2020-10-27:
I'm starting to refactor stage1.py such that the writing is separate from the calculations (creating a DL1Writer Component), and I've noticed that so far we have nearly everything needed in the event structure *except* the `MCDL1CameraContainer`, which is created on-the-fly inside the script, but is never written to the "event".  Now that I'm separating the computation from the writing, everything written must be in the event, so the question is: where would be a good place to put this info?

## Current status:

We have 2 containers for DL1 camera info. 
For Real data we have the following (in `event.dl1.tel[x]`):
```
ctapipe.containers.DL1CameraContainer:
                         image: Numpy array of camera image, after waveform
                                extraction.Shape: (n_pixel) as a 1-D array with
                                type float32
                     peak_time: Numpy array containing position of the peak of
                                the pulse as determined by the extractor. Shape:
                                (n_pixel) as a 1-D array with type float32
                    image_mask: Boolean numpy array where True means the pixel
                                has passed cleaning. Shape: (n_pixel) as a 1-D
                                array with type bool
                  parameters.*: Parameters derived from images
```

For simulated info we have a similar class (stored nowhere in the event structure, but created in the `Stage1ProcessorTool`):
```
ctapipe.containers.MCDL1CameraContainer:
             true_image: Numpy array of camera image in PE as simulated
                                before noise has been added. Shape: (n_pixel) as
                                a 1-D array with type float32
             true_parameters.*: Parameters derived from the true_image
```
 
we also have `event.mc.tel[x]`, which is:
```
ctapipe.containers.MCCameraEventContainer:
                    true_image: Numpy array of camera image in PE as simulated
                                before noise has been added. Shape: (n_pixel)
                      dc_to_pe: DC/PE calibration arrays from MC file
                      pedestal: pedestal calibration arrays from MC file
                   azimuth_raw: Raw azimuth angle [radians from N->E] for the
                                telescope
                  altitude_raw: Raw altitude angle [radians] for the telescope
                   azimuth_cor: the tracking Azimuth corrected for pointing
                                errors for the telescope
                  altitude_cor: the tracking Altitude corrected for pointing
                                errors for the telescope
```

## Options

1. get rid of `MCCameraEventContainer` and replace it with `MCDL1CameraContainer`: are any of the other fields still used? I think the pointing already goes in `event.pointing`, and `dc_to_pe` and `pedestal` are not really per-event quantities and are already in `event.calibration.tel[x].dl1` (`DL1CameraCalibrationContainer`). 

2. put MCDL1CameraContainer somewhere in `event.dl1.tel[x]` (a bit ugly, since this is only there for simulations)

Any others?

* `event.mc.tel[x]` right now contains

maxnoe, 2020-10-27:
I would go for 3:

Add a new `SimulationInfoContainer` at `event.simulation`,
that has all simulation info and that is `None` for observed data.

```
class SimulationInfoContainer(Container):
    run_header = Field(MCRunHeaderContainer) # rename to SimulationRunHeader?
    dl1 = Map(DL1SimulationContainer)
    ... any other fields
```

Get rid of MCCameraEventContainer, all the fields are now available somewhere else. (event.pointing or event.calib or event.simulation.dl1.tel[]...)

maxnoe, 2020-10-27:
#1180 could be solved in the same step

kosack, 2020-10-27:
We already removed the runheader from the event (it's not event-wise), so I think leaving that in the EventSource is better, but otherwise, I think this is an ok suggestion.  It's basically the same as  `event.mc` and replacing it with `event.simulation` however.

kosack, 2020-10-27:
I think I like `event.simulation.shower` best, since it mirrors `event.dl2.shower`

watsonjj, 2020-10-28:
>  dc_to_pe and pedestal are not really per-event quantities and are already in event.calibration.tel[x].dl1

No, these are not put in the `event.calibration` container. These are applied in SimtelEventSource as an R1 calibration.

kosack, 2020-11-02:
> EDIT: Ah from the PR I think you have realised this already

Yes, the question now is just do we want to store them somewhere? Right now I just apply them, but don't put them into the event structure anywhere.  Could be added back later in any case,  if we find a need.

PR #1495 (2020-11-03): Refactor simulation event info
Fixes #1493 

Major changes:
- removed `EventAndMonDataContainer` → merged with DataContainer  (and renamed to `ArrayEventContainer`)
- replaced `MCEventContainer` and `MCCameraEventContainer` with new classes: 
  + `SimulatedEventContainer`
  + `SimulatedShowerContainer`
  + `SimulatedCameraContainer`
- Removed `event.mc.*`, replaced with `event.simulation.*`
- `event.mc.tel[x].dc_to_pe` and `pedestals` from simulations are no longer stored in event...  should discuss if these should return and where (they are still applied however).  I would suggest if needed,  [... cut]
closed 2020-11-03 by kosack