cta-observatory/ctapipe #1497: developer install currently broken on macOS

kosack, 2020-10-28:
Using a clean install of miniconda (or anaconda), and following the install instructions fails for both python 3.7 and 3.8 with an error of `Found conflicts! Looking for incompatible packages.`.   Seems there is a missing package or too-strict dependency somewhere for macOS, since this works fine for Linux. 

We used to test installs on macOS in Travis - why did that stop?

maxnoe, 2020-10-28:
At some point the build did not work and we removed it.

kosack, 2020-10-28:
> At some point the build did not work and we removed it.

since many of the devs (myself included) use macOS, we should think about fixing that.

kosack, 2020-10-28:
Seems simply adding conda-forge to the channel list fixes it.  So I guess using `- conda-forge::<PACKAGE>` tries to get a single package from conda-forge, but not its dependencies

maxnoe, 2020-10-28:
@kosack that seems to be it.

By the way, I learned today that astropy imports their conda-forge packages to the astropy namespace, so that people don't have to solve vs. the full conda forge.

We could do the same.

maxnoe, 2020-10-28:
Wenn we add the `conda-forge` channel, now suddenly everything comes from forge. Do we want that?

kosack, 2020-10-28:
> By the way, I learned today that astropy imports their conda-forge packages to the astropy namespace, so that people don't have to solve vs. the full conda forge.

That's an option, since requiring everything from conda-forge is maybe not the best idea.  It means some more maintenance for us though, since I guess they have to move packages to their channel.

There must be some way to keep the basic packages in `defaults` like numpy MKL, and still use a few packages from conda-forge. Not sure why the deps are so restrictive now.

maxnoe, 2020-10-28:
For now you can probably simply add `conda-forge::corsikaio`, does that solve you problem?

maxnoe, 2021-02-12:
I think this was fixed? @kosack can you confirm?

commit 5669160a (2021-02-12): Add corsikaio, fixes #1497 (#1602)
closed 2021-02-12 by kosack