cta-observatory/ctapipe #1500: SimTelEventSource should provide access to histograms

kosack, 2020-10-30:
## Current Situation:
Currently in `stage1.py`, we use a hack to access the histogram information in the simtel file:
```py3
 hists = self.event_source.file_.histograms
```

For #1499 , this is not usable since we do not want to keep a reference to a parent event_source in the DL1Writer (to avoid config duplication), and also we should not be accessing private members in general

## Suggested change:

Provide the histograms as an attribute of `EventSource`, similar to the `mc_header` or `subarray` attributes. This could be quite low-level, not changing the format of the histograms for the moment (that can be done later as in `Stage1ProcessorTool.write_simulation_histograms()`. 
```py3
hists = source.simulation_histograms
print(hists[6]["n_bins_x"])
```

One problem is that I believe these are only readable when all events have been read already as they are stored at the end of the simtel file, so that makes this interface a bit ugly (and already stage1.py relies on calling this function only after all events are read).  So we would at least need to make this clear, like raising an exception if  the info is not yet available...

maxnoe, 2020-10-30:
The `_` is at the end. It is not a private variable, it's just to avoid the builtin name `file` ;)

But yes, a more direct way would be good. Probably directly exposing the `ShowerDistributionContainer` at the event source.

Which would then be the same for all event sources that produce simulated info.

kosack, 2020-10-30:
Yeah, I realized file_ isn't technically marked as private, but in a way it's exposing private information that is specific to the SimTelEventSource, so probably should be private! 

Really, the main problem was that if I require the user provides an EventSource in order to construct a DL1Writer, it poses a small problem: if I keep a reference to it, the config of DL1Writer will then include the  EventSource config  as a sub-config of DL1Writer (due to the removal of `add_subcomponent()`, which is not really correct.  In general it does seem strange to require an EventSource in order to write data (they could be generated randomly for example), rather than just passing in the relevant info. The only piece that I couldn't easily pass in was this histogram data, since it would have to work in a strange way. Another option would be to keep with storing the event_source reference, but make a way to have "weak" references (where the config is not "owned" by the parent). 

The only way I can think of making this work so far is a bit ugly: requiring the user to explicitly call it at the end of the event loop, rather than calling it in the `finish()` (or `__exit__()`) method:

```py3
with DL1Writer(...) as write_dl1:
    for event in source:
         ...
         write_dl1(event)
    write_dl1.write_simulation_histograms(source.file_)
```

maxnoe, 2020-10-30:
I think doing it outside the writer is fine

maxnoe, 2021-02-19:
I renamed this issue. We need *all* event sources to provide this info for simulated showers. For ctapipe this means both `SimTelEventSource` and `DL1EventSource`.

maxnoe, 2021-02-19:
We should add all these things to `EventSource` with default `None` or `empty` dicts and simulation event sources need to expose those.

maxnoe, 2023-10-21:
Duplicate of #1689

closed 2023-10-21 by maxnoe